sccn/eeglab #639: Advanced source localization

arnodelorme, 2023-06-26:
I'm a beginner EEGLAB user, I'm trying to follow the [manual](https://urldefense.com/v3/__https://eeglab.org/tutorials/09_source/EEG_sources.html__;!!Mih3wA!H4dDLJg14AcHS5nBa28FmZNU6Ij9aRLZS-twHMOgjJrVxG0k_yyrl1i9ecaIPVfzVqKb2XJkZqJLfVawbA_RGpOTAwQjmwinpabHJgD0vA$) in order to
practice source localization.

In particular I just tried to copy and paste the script described
about  the 'Advanced source reconstruction using DIPFIT/Fieldtrip'.

Check tutorial. There is a long email, but if the tutorial work it is ok.

arnodelorme, 2023-06-26:
@deepagupta will look into it.

deepagupta, 2023-06-28:
Hello,

Please note my comments below:

### Suggestions for implementation:

1.    Please consider adding somewhere that Fieldtrip-lite extension must be installed. If not, go to File-> manage EEGLAB extensions and install Fieldtrip-lite.
2.    Please note that while executing the code for figure “5:ft_sourceplot: sourceProj”, I get the following error message:

could not reshape "time" to the dimensions of the volume Warning: functional data is complex, taking absolute value Check for incorrect argument data type or missing argument in call to function 'abs'.  Error in ft_sourceplot (line 455)     fun = abs(fun);  Error in dipfit_example_tutorialtest20230823 (line 89) figure; ft_sourceplot(cfg, sourceProj);

3.    Please note that while executing the code for the section:
 %% Prepare leadfield surface
In the section ‘Performing source reconstruction:’
I get the warning:

Warning: adding C:\Users\d2gupta\Documents\eeglab2022.1\plugins\Fieldtrip-lite20230613\external\gifti toolbox to your MATLAB path Warning: use cfg.sourcemodel instead of cfg.grid

### Cosmetic changes’ suggestions to descriptive for clarity to the reader:
1.    Please consider changing the heading from "Equivalent dipole source localization of EEG data" to "Equivalent dipole source localization of EEG or ERP data"
2.    Positive affirmation first: Please consider changing first line to "EEGLAB provides a command-line implementation of the DIPFIT plugin to fit dipoles to raw ERP or EEG scalp maps that has otherwise not been expressly designed anywhere else including the plugin itself" from "Though the implementation of the DIPFIT plugin has not been expressly designed to fit dipoles to raw ERP or EEG scalp maps, EEGLAB provides a command-line function allowing DIPFIT to do so. "
3.    For giving the beginner an idea about what is DIPFIT as a starter, please consider hyperlinking DIPFIT plugin in the first line to: https://github.com/sccn/dipfit AND https://sccn.ucsd.edu/eeglab/dipfittut/dipfit.htmlold AND https://eeglab.org/tutorials/09_source/DIPFIT.html
4.    Please correct this: The “here” in “Click here to download the script above” is leading to an error message “Notice: Undefined variable: baseurl2 in /var/local/www/html/eeglab/locatefile.php on line 80 File not found”.
5.    Please include the eeglab_data_epochs_ica.set file or the instructions for how to fetch this sample dataset.
6.    In the section “Advanced source reconstruction using DIPFIT/Fieldtrip”:
a.    please make a subsection “History” or “Background” and then in that include the first paragrpah “DIPFIT relies on Fieldtrip, though in fact, DIPFIT was also an ancestor of Fieldtrip: when Robert Oostenveld, the first Fieldtrip developer, decided to release source imaging functions he had developed during his dissertation work, he first packaged them in EEGLAB as DIPFIT. A few years later, when he and his collaborators released Fieldtrip (also running on MATLAB), we reworked DIPFIT so it would use the Fieldtrip functions that Robert and colleagues planned to and have since maintained for use in Fieldtrip. Below is a short tutorial on how to perform source modeling using Fieldtrip applied to data in an EEGLAB dataset.”
b.    Also, one may consider hyperlinking here DIPFIT to the earlier links as highlighted in the above point#3 in this first paragraph of history.
c.    Subsection “Implementation” or “Performing source reconstruction in a volume” could include text from “First, use DIPFIT to align the electrode locations with a head model of choice (menu item Tools → Locate dipoles using DIPFIT → Head model and settings). The resulting DIPFIT information … and so on”
7.    Please correct this: script download link in the section ‘Performing source reconstruction on a surface’ is giving the error: “Notice: Undefined variable: baseurl2 in /var/local/www/html/eeglab/locatefile.php on line 80 File not found”
8.    Please consider perhaps specifying the following tutorial additionally in the relevant section: https://eeglab.org/tutorials/09_source/Model_Settings.html  and https://eeglab.org/tutorials/09_source/Custom_head_model.html

I hope the above is helpful
Thanking you
Sincerely
Deepa Gupta

arnodelorme, 2023-06-28:
Thanks Deepa. Can you issue a pull request on the tutorial. Let's start with the essential.

deepagupta, 2023-06-28:
Thank you Arnaud. Please see the created pull request here: https://github.com/sccn/eeglab/pull/643

arnodelorme, 2024-02-09:
Checked the tutorial and it is all good.

closed 2024-02-09 by arnodelorme