sccn/eeglab #647: IClabel GUI vs running from code given by eegh

dastout, 2023-07-05:
Hello,

I am attempting to use the IClabel feature and want to apply it to some of my (mostly) cleaned EEG resting-state data (6 minute eyes open). The data was mostly preprocessed using BVA (recorded using brain products). Using eeglab, I first used the eeglab GUI to run another bp filter (for a microstate analysis) and then run ICA and ICLabel to remove additional artifacts that were not accounted for in our BVA preprocessing stream (e.g., we already addressed eye-blink/saccades in BVA, but not muscle/head-movement,etc.). 

I first used the eeglab GUI to run through these steps ( bp filter 2-20Hz & ICA/ICALabel). Using the GUI I observed that after running ICA, IClabel identified 1 component as an artifact component (>.85). I then removed that component from the data. However, when I re-run this via the command line using the output from 'eegh' command line-by-line, I get a different number of components identified and then removed. I'm wondering what I'm doing wrong between the 2 versions. Here is the output of eegh that I ran exactly as shown. I'm wondering if 'eegh' didn't output the IC label selection?

**Output from eegh that I ran line-by-line:**
[ALLEEG EEG CURRENTSET ALLCOM] = eeglab;
EEG = pop_loadbv('/107/', '107_Resting_Segmentation_S3_0-6s.vhdr', [1 180000], [1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40 41 42 43 44 45 46 47 48 49 50 51 52 53 54 55 56 57 58 59 60 61 62 63 64 65 66 67 68 69]);
[ALLEEG EEG CURRENTSET] = pop_newset(ALLEEG, EEG, 0,'gui','off'); 
EEG = pop_eegfiltnew(EEG, 'locutoff',2,'hicutoff',20,'channels',{'Fp1','Fz','F3','F7','FT9','FC5','FC1','C3','T7','CP5','CP1','Pz','P3','P7','O1','Oz','O2','P4','P8','TP10','CP6','CP2','Cz','C4','T8','FT10','FC6','FC2','F4','F8','Fp2','AF7','AF3','AFz','F1','F5','FT7','FC3','C1','C5','TP7','CP3','P1','P5','PO7','PO3','POz','PO4','PO8','P6','P2','CPz','CP4','TP8','C6','C2','FC4','FT8','F6','AF8','AF4','F2','FCz','TP9'});
[ALLEEG EEG CURRENTSET] = pop_newset(ALLEEG, EEG, 1,'gui','off'); 
EEG = pop_runica(EEG, 'icatype', 'runica', 'extended',1,'interrupt','on');
[ALLEEG EEG] = eeg_store(ALLEEG, EEG, CURRENTSET);
EEG = pop_iclabel(EEG, 'default');
[ALLEEG EEG] = eeg_store(ALLEEG, EEG, CURRENTSET);
EEG = pop_icflag(EEG, [NaN NaN;0.85 1;0.85 1;0.85 1;0.85 1;0.85 1;NaN NaN]);
[ALLEEG EEG] = eeg_store(ALLEEG, EEG, CURRENTSET);
EEG = pop_subcomp( EEG, [], 0);
[ALLEEG EEG CURRENTSET] = pop_newset(ALLEEG, EEG, 2,'savenew','clean','gui','off'); 

**Data files for 1 subject:**
https://drive.google.com/file/d/1VLV-qfIECS_8JYpfstmczBDucUVVbvtJ/view?usp=drive_link

|   |   |
|----| ---- | 
| OS version     | MAC OS Ventura 13.4.1 |
| Matlab version | R2022a 9.12.0 |
| EEGLAB version | V2023.0 |

Thank you,

Danny

arnodelorme, 2024-02-09:
Every time you run the runica function, the results will be different because the random shuffling is different. rng(0) before running ICA to get deterministic results.

closed 2024-02-09 by arnodelorme

arnodelorme, 2024-02-09:
We will consider changing the default so results are reproducible.