sccn/eeglab #649: Problem with pop_dipfit_headmodel.m function

giupier95, 2023-07-06:
We wanted to create the head model by using our MRI. There is a problem with the function pop_dipfit_headmodel.m that you can see below. 

The command I'm using is:
pop_dipfit_headmodel(EEG, 'path/file.nii', 'nasion', [-3 4 0.5], 'lpa', [-1 -2 2], 'rpa', [.5 -.4 3])

Note: the coordinates are purely made up, they are just for testing. We would also like to know how to select the fiducials from the MRI before computing the head model.

The error is: 

Unrecognized function or variable 'fiducials'.

Error in pop_dipfit_headmodel (line 237)
fiducials = segmentedmri.transform * [ fiducials ones(size(fiducials,1),1)]';

You can find the MRI link here: https://drive.google.com/drive/folders/1x5ThYZmjZ3g0NBHrXlelGOjn2jXQeWr_?usp=drive_link

arnodelorme, 2023-07-06:
Thanks, I will look into it.

arnodelorme, 2023-08-08:
Thanks this was fixed in https://github.com/sccn/dipfit/commit/0669b0d87b3afc9cd3c1924248bf6dea725a8f76

closed 2023-08-08 by arnodelorme