sccn/eeglab #652: Help epoching

kr1st1na7, 2023-07-08:
How can I epoch with start and end set by two event marker instead of a time surrounding one marker?

kr1st1na7, 2023-07-13:
> In EEGLAB you cannot extract epochs of different length. You can extract
> data segments if you write a script.

Hi,
Could you clarify what you mean by extracting data segments if I write a
script? Are there any steps available?

carrie372, 2024-01-18:
I used custom scripts to do variable epoching from -200-ms from start of trial to time of response. However, the eeglab interface is not registering this information (epochs not updating when running "eeglab redraw"), trials are not recognized, and plotting functions no longer work. My coding is correct in terms of pulling events, calculating the epoch duration, etc.

Any ideas on how to fix this? Attached are outputs of EEG.event and EEG.epochs after running the attached code.

[image]
[image]
[image]