sccn/eeglab #653: inspect/label component by map

taliehmir, 2023-07-09:
### Description

[i used to work with the option 'inspect/label component by map' and click on each component to see the component with plot, suddenly that doesn't work appropriately any more i get always error and ERP plot is empty.]

https://drive.google.com/drive/folders/1XpjN8M3AFlglnv7qgCXH6EmbTFJxVtWH?usp=sharing

#### Steps to Reproduce

1. [eeglab]
2. [file> load existing dataset]
3. [inspect/label component by map> IC1]

<br>
#### Expected behavior:
[component with 3 plots]

<br>
#### Actual behavior:
[shows only two plots, ERP plot is empty]

<br>
#### Versions

| OS version | macBook Air |
| Matlab version | R2023a |
| EEGLAB version | 2023.0 |

arnodelorme, 2023-07-13:
It works for me with your dataset. Can you try with the EEGLAB version on Github? 
[image]

closed 2023-07-13 by arnodelorme

taliehmir, 2023-08-17:
I have already tried with the EEGLAB version on Github, but it didn't work
again!