sccn/eeglab #657: "mexsload not found in path" Issue

alparslanonder, 2023-07-17:
Hi, While I was trying to import my edf file (EDF-C) I faced with this problem. I removed then loaded the latest version of Biosig, try lots of things but I keep having this problem, how can I solve it?

Thanks a lot.

[image]

arnodelorme, 2023-08-08:
Do you have windows. If this is the case, then mexSLOAD should be installed. If you do not have windows then it is more complex. https://sourceforge.net/p/biosig/wiki/FAQ/

arnodelorme, 2024-02-09:
Closing this for now. I updated the help message.

lopezchau, 2024-05-04:
> Do you have windows. If this is the case, then mexSLOAD should be installed. If you do not have windows then it is more complex. https://sourceforge.net/p/biosig/wiki/FAQ/

Dear Arno, I have a MacOS system and followed the steps but I was not able to open an EDF file with mexSLOAD.  It is of my interest because events are not loaded with the default settings.

arnodelorme, 2024-05-04:
You do not need mexsload to load EDF events. You can use other options when calling the biosig function. For example, try importing with annotation off.

```
EEG = pop_biosig('file.edf', 'importannot', 'off');
[ALLEEG EEG CURRENTSET] = eeg_store(ALLEEG, EEG);
eeglab redraw
```

lopezchau, 2024-05-04:
> You do not need mexsload to load EDF events. You can use other options when calling the biosig function. For example, try importing with annotation off.
> 
> ```
> EEG = pop_biosig('file.edf', 'importannot', 'off');
> [ALLEEG EEG CURRENTSET] = eeg_store(ALLEEG, EEG);
> eeglab redraw
> ```

Thank you so much for your answer!  Still no luck while importing events.  When I try other import plugins like import_edf the events are imported normally but I get an error that says there is another edfread function and I need to remove it from the path.

Sorry if the question is too basic, but these are my first steps with EEGLab.

[image]

arnodelorme, 2024-05-04:
Can you share your EDF file?

lopezchau, 2024-05-04:
> Can you share your EDF file?

Absolutely Arno. It would be such an honor.  I am new to EEGLab and having support from you is unbelievable.

Thank you so much.  Greetings from Peru.

PS: Just to let you know that data have being acquired from a Neuron Spectrum 64. It is a russian made amplifier. We are planning to do some research with it and a same brand TMS device (TMS-EEG).

https://neurosoft.com/en/catalog/eeg/neuron-spectrum-64#overview

EDF File:
https://drive.google.com/file/d/1--m7MZvoHEpzg_sh_gTh_adeqhgh22Ou/view?usp=sharing

Lucreziadimarino, 2024-10-20:
@lopezchau Ciao! Ho lo stesso identico problema... hai risolto?

alparslanonder, 2024-10-20:
As I experienced, the exact problem was in the Headers of the .edf files, with a pipeline I solved the issue about the headers than I did have some issue again.

Lucreziadimarino, 2024-10-20:
che tipo di problemi? perchè io non riesco in alcun modo a visualizzare gli eventi nonostante io abbia la certezza che ci siano. Sono una studentessa che sta eseguendo un progetto sperimentale per la tesi e non sono riuscita ancora a risolvere

closed 2024-11-11 by arnodelorme