sccn/eeglab #659: welch() at line 87: Incorrect use of zeros ; Size The value must be an integer

a1phaeeg, 2023-07-22:
### Description

[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]
When I just  want to plot→channel spectra and maps，it always reports errors: EEGLAB  error in function welch() at line 87: 
Incorrect use of zeros 
Size The value must be an integer

The file has been uploaded to Google drive, as shown below 
https://drive.google.com/drive/folders/1QnhZySVRuG6oAD3vzw3zyvZ8VMQP5kbI?usp=sharing

The EEG data *.set was converted from *.edf files.
[image]

<br>
#### Expected behavior:
[*help!help!help! please...*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*2020a*] |
| EEGLAB version | [*2022.1*] |

a1phaeeg, 2023-07-22:
If anybody has any method, please let me know, I would like to try.Thank you!

arnodelorme, 2023-08-08:
I have requested access to the file.

a1phaeeg, 2023-08-16:
> I have requested access to the file.

Mr. Delorme, thank you for response. Please forgive me if I didn't have permission to open Google drive before, but I do now.

a1phaeeg, 2023-08-21:
> I have requested access to the file.

Hi, Mr. Delorme. 
I've authorized the files from Google Drive to you.
Looking forward  to your reply, thanks again!

closed 2023-10-20 by a1phaeeg