sccn/eeglab #660: Java exception

vmathew7, 2023-07-24:
Hi
I am a Ph.D. scholar working on face processing studies. We use EGI 65 electrode system in our lab. While trying to analyse data using EEGLAB we used the latest version of MFFimport4.1. 
When I am using MFFMatlabIO to import files I am getting the following Java exception error, which I have attached as an image.
This is only happening to few files. 
Please help about how to import the file for doing analysis.

Sincerely
Thanking you in advance

[image]

vmathew7, 2023-07-25:
Thank you for the response. I am sharing the link for accessing the file in your Email, hope that's fine. https://drive.google.com/drive/folders/1FVYPbmbK1slWDj_biN3w4MwPlYTK5GOZ?usp=drive_link

arnodelorme, 2023-07-25:
Yes, thanks

vmathew7, 2023-07-28:
Hi, hope the file was accessible. I just wanted to follow up on this issue. Is there something I can refer to identify the issues and make changes to import the files?

Thanking you in advance

arnodelorme, 2023-08-08:
Note to self: removing the channels n="" solves the issue. A fix would be to edit the file and remove such channels.

vmathew7, 2023-08-09:
Thank you so much, I have made the changes by deleting n="" channels in the info1.xml file and it's importing. Thank you again for your time.

arnodelorme, 2024-02-09:
Fixed.

closed 2024-02-09 by arnodelorme