sccn/eeglab #661: EEGLAB error in function loadcnt() at line 558:

HingHuang, 2023-07-26:
### Description

When I import a file in .CNT format, an error is reported that EEGLAB error in function loadcnt() at line 558: The assignment cannot be made because the size on the left is 10456×963182912 and the size on the right is 1×38783421.
[image]

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Windows 11*] |
| Matlab version | [*MATLAB R2019a*] |
| EEGLAB version | [*EEGLAB v2023.0*] |

arnodelorme, 2023-07-26:
Can you share a file?

HingHuang, 2023-07-26:
>  Can you share a file?
>
>  &gt;
>  &gt;
>  &gt; Description
>  &gt;
>  &gt; When I import a file in .CNT format, an error is reported that EEGLAB error in function loadcnt() at line 558: The assignment cannot be made because the size on the left is 10456×963182912 and the size on the right is 1×38783421.
>  &gt;
>  &gt;
>  &gt;
>  &gt; #### Steps to Reproduce
>  &gt; 	• [First Step]
>  &gt; 	• [Second Step]
>  &gt; 	• [and so on...]
>  &gt;
>  &gt; #### Expected behavior: [*What you expected to happen*]
>  &gt; #### Actual behavior: [*What actually happened*]
>  &gt; #### Versions
>  &gt; OS version	[Windows 11]
>  &gt; Matlab version	[MATLAB R2019a]
>  &gt; EEGLAB version	[EEGLAB v2023.0]
>  &gt; —
>  &gt;
>
>
>
> 从QQ邮箱发来的超大附件
>
> LNE_01_2023-06-02_21-17-41.cnt (88.31M, 2023年08月25日 12:07 到期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?t=exs_ftn_download&k=6135333802384ace65a5ae6e1437031b4a410a08530301561f060a0b541a090200531e0b005207190502570f0101070103570b0d322931787c706c080368030400061e08041a01066d07021503001c00031b505646370c&code=25382714

?机长
***@***.***

arnodelorme, 2023-08-08:
This is because this is ANT file, not a Neuroscan file. Use the "ANTeepimport" v1.14 EEGLAB plugin to import this file. I imported it myself successfully.

closed 2023-08-08 by arnodelorme