sccn/eeglab #664: Bug 'chantype' on BRAINBEATS toolbox for EEGLAB

damiana94, 2023-08-07:
Hi! I was trying to use your new toolbox but there seems to be a bug when you try to upload the data (even the sample data are not usable). 

The error says: 
pop_select error: undefined argument 'chantype'

However, even when EEG.channels.type is filled is, the error continues to appear.

arnodelorme, 2023-08-08:
I think this is because you are using an old version of EEGLAB. @amisepa what do you think?

closed 2023-08-08 by arnodelorme

damiberg, 2023-08-08:
the eeglab version I am using is 2022.1, not so old.

amisepa, 2023-08-10:
What do you mean by "upload" the data?

damiberg, 2023-08-10:
No, I didn't manage to solve it. Basically I selected your data to try, i go to Tools>Brain Beats> Process file (subject level). I choose the channel (ECG one) and the analysis to do and so on and then I press OK, but it gives me this error (line 220 of pop_select)

amisepa, 2023-08-10:
Did you load the data into EEGLAB first? 
File > load an existing dataset > go to sample_data folder in the BrainBeats folder and select the corresponding .set file. 

I am noticing there is a mistake in the preprint. Step 1.1 describes this loading step, but the figure shows the next step to launch the processing. Sorry if that was misleading. We'll make sure it's corrected during the next revision.

damiberg, 2023-08-10:
Yes of course. Maybe is really due to the version of eeglab I am using?

amisepa, 2023-08-10:
Can you please provide a little more detail? 
Did you install BeainBeats via eeglab extension manager or via GitHub. Which sample_dataset and which method did you try? 

I'm getting another error from extension manager, which I solved in the newer version in GitHub. I tried HEP and features on sample_data1.set and it worked fine.

amisepa, 2023-08-11:
I'll take a closer look into your issue soon