sccn/eeglab #667: Error with starting process of Re-referencing data

MatteoDaBergamo, 2023-08-16:
### Description

Trying to re-reference data, but getting error when trying to edit channel info. 

<br>
#### Steps to Reproduce

1. File -> Load existing dataset -> test00000x_noise.set
2. Edit -> Channel Locations
3. use MNI coordinate file for BEM dipfit model
4. At this point, when the pop_chanedit() screen pops up, I'm unable to go through any of the other channels (stuck on 1). any attempts to change channels results in an error.

<br>
#### Expected behavior:
Expecting to go to channel 63 and append a channel to use as a reference during re-referencing, but unable to do anything.

<br>
#### Actual behavior:
When trying to cycle through channels, receive error message:

Brace indexing is not supported for variables of this type.

Error in pop_chanedit (line 1033)
    userdata.commands = { userdata.commands{:} args{:} };
 
Error while evaluating UIControl Callback.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | Windows 10, OS Build 18363.2274 |
| Matlab version | R2022b version 9.13 |
| EEGLAB version | latest version (just did clean install this morning) |

#### Data file I'm using
[test00000x_noise.zip](https://github.com/sccn/eeglab/files/12360898/test00000x_noise.zip)

AminAlam, 2023-10-21:
You have only uploaded .set file. Please upload .fdt file as well.

arnodelorme, 2024-02-09:
I was able to import the file, but could not reproduce the error. I was able to add a reference channels and then rereference to that channel.

closed 2024-02-09 by arnodelorme