sccn/eeglab #668: edfannot2evt error

kunnnnnnnn, 2023-08-19:
### Description

Hello，I try to import data from bdf file and this error happens：

EEGLAB error in function edfannot2evt() at line 52

Cannot perform assignment because the index on the left is not compatible with the size on the right 

<br>
#### Steps to Reproduce

1. [import data from biosemi bdf file]
2. [error]

<br>
#### matlaB 2023a  eeglab 2023.1

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |
[20230818202543_1.zip](https://github.com/sccn/eeglab/files/12385621/20230818202543_1.zip)

arnodelorme, 2023-10-24:
Can you share the data?

arnodelorme, 2024-02-09:
@schloegl this is a BIOSIG issue. Would you mind looking into this. The file is attached in a ZIP.

schloegl, 2024-02-09:
I checked this file, and I can reproduce the issue when using sopen('20230818202543_1.bdf'). I can suggest the following three workarounds. 

1.  apply the following patch to sopen.m, it disables the conversion of the edfplus annotations into the event table: 
```
diff --git a/biosig4matlab/t200_FileAccess/sopen.m b/biosig4matlab/t200_FileAccess/sopen.m
index d293e309..c574ebab 100644
--- a/biosig4matlab/t200_FileAccess/sopen.m
+++ b/biosig4matlab/t200_FileAccess/sopen.m
@@ -1054,7 +1054,6 @@ end;
                         
                 end;
                 HDR.EVENT.SampleRate = HDR.SampleRate;
-               HDR = edfannot2evt(HDR);
 
                 status = fseek(HDR.FILE.FID, HDR.HeadLen, 'bof');
                 HDR.FILE.POS  = 0;

```

2) try getting mexSLOAD.mex in place for your platform. This can handle that data. 

3) Another workaround is to convert the data with biosig's "save2gdf" command into gdf-format and then loading that file into matlab/eeglab. This will also provide the correct event table.

I hope this helps.

arnodelorme, 2024-02-09:
Thanks @schloegl , I have clarified the error message in EEGLAB so people give it a shot with mexSLOAD (also available in EEGLAB for Windows -- OSx and Ubuntu requires to go to your BIOSIG FAQ).

closed 2024-02-09 by arnodelorme