sccn/eeglab #672: Clustering plot error with EEGLAB 2023.0 version

deepagupta, 2023-09-01:
### Description

Clustering plotting bug as described below with screenshots attached

<br>
#### Steps to Reproduce

1. Load your datasets
2. Create STUDY with all the loaded datasets (selected r.v. <15% and removed external cortical dipoles)
3. Performed clustering (Study -> PCA clustering(original) ->'Build preclustering array' (dipoles weightage 1) followed by the 'cluster components' (outliers checked with std. dev. 3 and k=27) step. 
4. Plot/edit - > plot dipoles option gives error as can be seen in the attached screenshot

<br>
#### Expected behavior:
dipole plots (attached by using EEGLAB 2022.0 on the same windows machine and MATLAB 2022a works fine as shown in the screenshot below)

[image]

<br>
#### Actual behavior:
error message with incomplete plot as can be seen below (attached by using EEGLAB 2023.0 with same matlab and windows version)

[image]
[image]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Windows] |
| Matlab version | [2022a] |
| EEGLAB version | [2023.0] |

arnodelorme, 2023-09-01:
Can you type

which dipplot

Are you using dipfit 5.3?

deepagupta, 2023-09-06:
Thank you Arnaud. That helped, however, with ERPimage, I still get the following error message. Please advise:

[image]

On investigating, I am seeing that the array at that line#243 is empty. I checked, all my plugins are now up to date but I am still getting the above error message on visualization. If possible, may I request some hints on how to proceed better please? Thank you in advance

MATLAB: 2022a
EEGLAB: v2023.0

closed 2023-09-09 by deepagupta