sccn/eeglab #677: error in function pop_writebva()

cg091, 2023-09-19:
Hello,

I am attempting to convert severals .set EEG processed files into Brainvision format, using the function pop_writebva().

However, for two files in particular I receive the following error:
[image]

I tried to update the bva-io plugin, and also try to use the GUI but nothing changed.

Do you have any idea why this is happening?

Thank you very much

arnodelorme, 2023-09-19:
Would you mind to submit your dataset?

arnodelorme, 2023-09-19:
@nucleuscub you mentioned that maybe someone could assist with supporting that plugin.

cg091, 2023-09-19:
@arnodelorme 
thank you for your answer. 
Here is the link for the dataset, because I can't upload files larger than 25MB:
https://drive.google.com/file/d/19A3e9wdzQz2NR8UFf6rZ9fCLl6AI85ji/view?usp=drive_link

arnodelorme, 2024-02-08:
Apologies for the late answer. The dataset is corrupted with more than 1 latency per event which is not theoretically possible. It would be interesting to see how this dataset came to be in EEGLAB (wether the event structure was automatically imported or wether it was corrupted by a custom program).

I am closing the report for now.

closed 2024-02-08 by arnodelorme