sccn/eeglab #678: Importing .eeg file

sodeleoz, 2023-09-21:
Hello,
I'm new to using this and I've been trying to import .eeg files to preprocess however once the I try to leave the spaces for the subsets blank, I receive this error.
[image]
When I do fill out the blanks, I receive this error.
[image]
How may I fix this? 
And is there another option to be able to import all of the .eeg files at the same time? Creating a study only allows .set.

Here is the dataset I am using:
https://drive.google.com/drive/folders/1M9wNaskVQqI3ZGQVdgvA3YHZp72MQOsy?usp=sharing

Thank you.

AminAlam, 2023-10-21:
I think your datasets have a problem, as the problem is not because of EEGLAB. The error is coming from neuroscanio1.7 plugin and the reason for this errors is that your .eeg datasets don't have any channels (chan= fread(fid, 1, 'ushort') is an empty list)

arnodelorme, 2024-02-09:
Agreed, you should use the BVA-IO plugin. These are brain vision exchange files.

closed 2024-02-09 by arnodelorme