sccn/eeglab #680: pop_writeeeg modifies event codes

Shervin00, 2023-09-28:
### Description
Hi! I am attempting to merge two .bdf-files recorded during the same cognitive task using pop_mergeset and subsequently pop_writeeeg to save the merged files as one new bdf. However, while doing so I noticed that the output of pop_writeeeg does not have an event structure that matches that of the original files. The trigger codes stored in EEG.event are modified.

This happens to files saved with pop_writeeeg irrespective of pop_mergeset being used prior or not.
Is there any way to fix this?

[image]
[image]

#### Steps to Reproduce
1. [Load biosemi data file using pop_biosig]
2. [Use pop_writeeeg to save the file as a new bdf file]
4. [Load the new file with pop_biosig]
3. [Compare their EEG.event]
5. Despair

<br>
#### Expected behavior:
[That the contents of the event structure remains identical to input.]

<br>
#### Actual behavior:
[The event structure is modified.]

<br>
#### Versions
|   |   |
|----| ---- | 
| OS version     | [Red Hat Enterprise Linux, Linux 3.10.0] |
| Matlab version | [Matlab R2020b] |
| EEGLAB version | [EEGLAB 2022.0 and EEGLAB 2023.] |

Shervin00, 2023-09-29:
Hi Arno,
Thanks for your swift reply! However, I am not entirely sure which option you are referring to. I tried opening the output file from pop_writeeeg with various different options in pop_biosig(), including all varieties of 'bdfevenmode', 'importevent', 'importannot', 'importmex', 'overflow' settings with no success.

arnodelorme, 2023-09-29:
I believe it is this one

```
   'bdfeventmode' - [integer] see bdf2biosig_events function help. Default is 4.
```

Shervin00, 2023-10-02:
Thanks. Changing the 'bdfeventmode' setting didn't work unfortunately (I tried all integers)

Nathan-Thomas-Han, 2023-12-23:
Hi, I just wanted to ask if you managed to find a solution to this? I had a similar issue where when I exported to BDF, all of my trigger values changed to 31 (from 251, 252, and 253).

arnodelorme, 2023-12-23:
I have tried it in the past changing 'bdfeventmode' and it worked. Would you mind sharing your BDF file?

Nathan-Thomas-Han, 2023-12-23:
Thanks for replying so quickly, I've included the BDF file here:

https://drive.google.com/file/d/15Wc1t_16LJv1EzoMgFVF7XMpWKMGgZNB/view?usp=sharing

arnodelorme, 2023-12-23:
This is a different issue. You mean you exported from EEGLAB to BDF. Do you have the original EEGLAB file?

arnodelorme, 2023-12-23:
> Thanks. Changing the 'bdfeventmode' setting didn't work unfortunately (I tried all integers)

Try disabling Import EDF+ events.

Nathan-Thomas-Han, 2023-12-23:
https://drive.google.com/drive/folders/1opmx-kiFZYbu4djZNZQO-E0xZtOwOG52?usp=sharing

Hi, here is the folder with the EEGLAB dataset. For context, the 252 trigger was originally 30ms ahead of where it should have been so I manually shifted it back -30ms by first exporting the events via a text file then imported it back after manually changing the latency values. The dataset I've included is with the updated event markers.

arnodelorme, 2024-02-09:
Closing this for now.

closed 2024-02-09 by arnodelorme