sccn/eeglab #682: pop_readbdf error

pb865, 2023-10-06:
I am creating a pipeline to process from EEG file sin .bdf format.

When I use the EEGlab GUI (Import data --> Using EEGlab functions and plugins -->From Biosemi toolbox --> File), it works fine. When I try to write a line of code to import the file, it returns an error.

Right now I have one EEG file (called pilot01.bdf), for which I have some code that will read in the subject IDs into a list called SUB. It is able to recognize SUB{i} = 01. after I specify i = 1. 

>>eeglab;

>>DIR = ("C:\Users\filepath\EEGFiles");
SUB = {};
x = dir(DIR);
for y = 1:length(x)
    out = regexp(x(y).name, '\d*.bdf');
    if (~isempty(out))
        SUB = [SUB,x(y).name(6:end-4)];
    end
end

>> i = 1

 >> [ALLEEG EEG CURRENTSET ALLCOM] = eeglab;

>> EEG = pop_readbdf([DIR '\pilot' SUB{i} '.bdf']);
Reading BDF data in 24-bit format...
Unable to perform assignment because dot indexing is not supported for variables of this type.

>>Error in sopen (line 83)
        HDR.FILE.stdout = 1;

>>Error in pop_readbdf (line 211)
dat = sopen(filename);

Does anyone know what these errors mean? Why would it be able to read in the file in the GUI but not via the pop_readbdf() function?

Thanks in advance! Let me know if I have not been clear on anything.

arnodelorme, 2023-10-06:
Can you try with BIOSIG instead?

pb865, 2023-10-06:
> Can you try with BIOSIG instead?

Sorry, I'm not sure what you mean? This function is in the BIOSIG toolbox from my understanding, so I should already have it in my EEGlab folder and added to the path in MATLAB.

arnodelorme, 2023-10-06:
Import the BDF with BIOSIG (menu item File > Import Data > Using BIOSIG)

pb865, 2023-10-06:
> Import the BDF with BIOSIG (menu item File > Import Data > Using BIOSIG)
> […](#)

Yes, it works in the GUI using this method. The issue is I would like to have it as a line of code so that I can process multiple people at once using a loop in a pre-written script. For some reason, it throws errors when I use the code pop_readbdf() function.

arnodelorme, 2023-10-06:
You can also use the pop_biosig function from the command line. You can always share your file pilot.bdf and we can have a look (probably not soon since there is a workaround) as to why the pop_readbdf function fails.

pb865, 2023-10-06:
> pop_biosig

Thanks! 

I just tried pop_biosig. I get similar errors.

>>     EEG = pop_biosig([DIR '\pilot' SUB{i} '.bdf']);
sopen mode is "OVERFLOWDETECTION:OFF"
Unable to perform assignment because dot indexing is not supported for variables of this type.

>>Error in sopen (line 83)
        HDR.FILE.stdout = 1;

>>Error in pop_biosig>readfile (line 324)
dat = sopen(filename, 'r', channels, strmode);

>>Error in pop_biosig (line 207)
    [dat, DAT, interval] = readfile(filename{iFile}, g.channels, g.blockrange, g.memorymapped, g.bdfeventmode, g.overflow, g.uncalibrated);

arnodelorme, 2023-10-06:
The file might be corrupted. Can you try loading it with another tool?

pb865, 2023-10-06:
> The file might be corrupted. Can you try loading it with another tool?

It still loads fine when I use the GUI (menu item File > Import Data > Using BIOSIG).

arnodelorme, 2023-10-06:
Ah, then you can use the call present in the EEGLAB history. Type

```
eegh
```

On the MATLAB command line to see how EEGLAB reads your file.

pb865, 2023-10-06:
> Ah, then you can use the call present in the EEGLAB history. Type
> 
> ```
> eegh
> ```
> 
> On the MATLAB command line to see how EEGLAB reads your file.

Ah, Thank you! For some reason replacing the DIR shortcut with the direct file path given by the eegh command output worked.

closed 2023-10-06 by arnodelorme

arnodelorme, 2023-10-06:
Great!