sccn/eeglab #687: incompatibility between erplab (10 or 8.3 or 9.0) and eegab 2023.1 for simple voltage threshold in erplab

clreed, 2023-10-13:
### Description
Hi,
We were running the artifact detection simple voltage threshold function on erplab10.0 and eeblab2023.1.
Although the moving window works, there is an error running the AR threshold function from the GUI.  I have tried several version of erplab with the 2023 eeglab (running matlab 2023a)and they all crash.  It seems like something is amiss.

Error is:
EEGLAB error in function isFieldNested() at line 196:
Index 3xceeds the number of array elements. Index must not exceed 0.

This function works with other versions of eegalb and matlab, but not eeglab2023
Seems to be a bug. Do you know how to fix the code to make the two programs compatible?

to reproduce the error, open the dataset that has been filtered, reref'd, baseline corrected and epoched.
use the erplab gui and select Artifact Detection
select simple voltage threshold
the various entries in for the epoch, the threshold, which electrodes etc don't make any difference
they all produce the same error.

Note: this same function works with erplab all versions with eeglab 2021, eeglab 2022 but not eeglab 2023.

Cathy Reed (clree@cmc.edu)
[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |

arnodelorme, 2023-10-14:
Would you mind posting to the ERPLAB github instead https://github.com/ucdavis/erplab

closed 2023-10-14 by arnodelorme

clreed, 2023-10-14:
I did. It seems to be an issue with the latest eeglab version

-------------------------------
Catherine L. Reed, Ph.D.
McElwee Family Professor of Psychology & George R. Roberts Fellow
Psychology & Neuroscience
Claremont McKenna College Claremont CA 91711
Phone: (909) 607-0740
Email: ***@***.***
Web site: http://www1.cmc.edu/pages/faculty/clreed/

arnodelorme, 2023-10-14:
If they cannot find the problem, send me us a data test and instructions
on how to reproduce the problem. Thanks

clreed, 2023-10-14:
HI,
So the issue seems to be something related to the event structure in the eeglab set file-- this only happens with eeglab 2023, not 2022 or 2021, or earlier.  Do you know what might have changed? The reason why I am asking is that we have 300 person database with each person completing the same six experiments.  We are creating an open access EEG/ERP course and I really want our dataset to be compatible with the latest versions of eeglab and with our course materials.  The data sets that have this problem have 4 bins-- condition 1 correct, cond2 correct, cond1 incorrect, and cond2 incorrect.  Bins 3 and 4 often do not have any trials in them because participants did not make errors.  I know that that this could be an erplab issue but given that the files have been used extensively with previous versions and the issue comes up with loading the set file, I was hoping you might have some ideas.
It is always ok for us to just tell people not to update to eeglab2023 but that I not optimal for anyone.

This is the error I get when loading the files that won't do the AR threshold but will do the AR peak to peak window:
pop_loadset(): loading file Z:\Summer_RA_Data\PURSUE N400\n400_files_to_process_to_ave\1054\1054_N400_epoch.set ...
Warning: Inconsistency between urevent (backup) and event structures, removing urevent structure

Error from AR simple voltage threshold in erplab:

EEGLAB error in function isFieldNested() at line 196:
Index 3xceeds the number of array elements. Index must not exceed 0.

This function works with other versions of eegalb and matlab, but not
eeglab2023

Thanks, Cathy Reed
-----------------------------------
Catherine L. Reed, Ph.D.
McElwee Family Professor of Psychology & George R. Roberts Fellow
Professor of Psychology & Neuroscience
Psychological Science, Claremont McKenna College
Mailing Address: 850 Columbia Ave, Kravis 108, Claremont CA 91711
Phone: (909) 607-0740
Email: ***@***.******@***.***>
Web site: <https://pages.cmc.edu/clreed/> http://www1.cmc.edu/pages/faculty/clreed/

This e-mail from ***@***.*** was generated by an EXTERNAL email server
 out-27.smtp.github.com               (  Arnaud Delorme ***@***.***> )

arnodelorme, 2023-10-15:
Could you share a dataset and the exact procedure to reproduce the problem?

clreed, 2023-10-15:
Hi
I just sent 2 emails with files to you. Appreciate your help
Cathy

-------------------------------
Catherine L. Reed, Ph.D.
McElwee Family Professor of Psychology & George R. Roberts Fellow
Psychology & Neuroscience
Claremont McKenna College Claremont CA 91711
Phone: (909) 607-0740
Email: ***@***.***
Web site: http://www1.cmc.edu/pages/faculty/clreed/

 out-26.smtp.github.com               (  Arnaud Delorme ***@***.***> )

clreed, 2023-10-24:
Hi Arnaud,
I just wanted thank you for hearing me out regarding the eeglab2023 + erplap10.0 mismatch.
David Garrett from ERPlab was able to fix the artifact rejection procedures in erplab and now has an updated erplab 10.1 that does not produce the error.  Apparently it had nothing to do with my BDF file or anything with the previous file processing.

However, I was contacting you because even if I created the file using eeglab2023 it would give me the loading error about the event structure.
The good news is that according to David Garrett, the error message is one of those alert messages but is not catastrophic to processing.
Phew.

As I said, we have a PURSUE (pursueerp.com) database that has 6 experiments and 300+ participants who have files like this and we are creating undergraduate level teaching materials so it was critical to know that the latest versions of the eeglab and erplab were compatible with the datasets.

Best, Cathy Reed

-----------------------------------
Catherine L. Reed, Ph.D.
McElwee Family Professor of Psychology and George R. Roberts Fellow
Professor of Psychology & Neuroscience
Dept of Psychology, Claremont McKenna College
850 Columbia Ave, Claremont CA 91711
Phone: (909) 607-0740   Email: ***@***.******@***.***>
Website: https://www.cmc.edu/academic/faculty/profile/catherine-reed