sccn/eeglab #688: Query Regarding Repeated Measures ANOVA in LIMO 4.0 and EEGLAB 2023.1

npdrbong, 2023-10-17:
### Description

[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]

Dear Cyril Pernet and Arnauld Delorme,

First and foremost, thank you for creating such invaluable tools. I am currently conducting an experiment where I aim to compare 3 different conditions based on events occurring within specific blocks.

I am utilizing LIMO 4.0 within EEGLAB 2023.1 for this analysis. In my STUDY, there are a total of 5 event markers which I have categorized into three groups: Fix, OMS&pump&stop, and Cashout&pop. I intend to perform a Repeated Measures ANOVA on data from 42 participants to examine any differences between these conditions. As my work is closely aligned with the tutorials provided, I have been following the tutorial steps:

Precompute
Run ERP calculations in pop_limo() with the following parameters:
pop_limo(STUDY, ALLEEG, 'method','WLS','measure','daterp','timelim',[-150 650] ,'erase','on','splitreg','off');
Proceed to 2nd level analysis -> check on 'Precompute statistics' -> Select ANOVA/ANCOVA -> Repeated Measures ANOVA -> 'Continue' for group information (only one group) -> Enter repeated factors level {Factor 1 : BART(name) : 3(Number of values)} -> Beta -> pick a 1st level analysis file 'ok' -> pick one or more beta parameters below 'select 1~6(except constant)'
After performing these steps, I encounter the following error message:
```
Index exceeds the number of array elements. Index must not exceed 1.

Error in limo_random_select>reorganize_params (line 2304)
str = ['Reorganize values as needed: ' factorname{1} '=rows and ' factorname{2} '=columns'];

Error in limo_random_select (line 1335)
                    paramTmp        = reorganize_params(parameters(i,:), betas,factor_names, factor_nb);

Error in limo_random_effect>ANOVA_Callback (line 262)
            limo_random_select(answer,handles.chan_file,'nboot',handles.b,'tfce',handles.tfce,'type','Channels');

Error in gui_mainfcn (line 95)
        feval(varargin{:});

Error in limo_random_effect (line 36)
    gui_mainfcn(gui_State, varargin{:});

Error in matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)limo_random_effect('ANOVA_Callback',hObject,eventdata,guidata(hObject))
 
Error while evaluating UIControl Callback.
```

<br>
#### Expected behavior:
[*What you expected to happen*]
[image]

<br>
#### Actual behavior:
[*What actually happened*]
[image]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | window10.0
| Matlab version | 2022a
| EEGLAB version | 2023.1 |

Could the issue stem from incorrect input regarding the number of factors or beta parameters? I would appreciate your insights into resolving this matter.

Best regards,
Bong

npdrbong, 2023-10-17:
I am sorry. I done this with selecting each con file instead of beta.

arnodelorme, 2023-10-17:
Would you mind to share your study so we can try running the commands and see what's wrong? thanks

CPernet, 2023-10-18:
it's the factor name issue we have fixed it in this version of [limo_random_select](https://github.com/LIMO-EEG-Toolbox/limo_tools/blob/master/limo_random_select.m)

arnodelorme, 2023-10-24:
OK, closing for now. Feel free to reopen.

closed 2023-10-24 by arnodelorme