sccn/eeglab #690: welch() at line 87: Incorrect use of zeros ; The value of Size must be an integer

a1phaeeg, 2023-10-20:
### Description
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.]
When I just want to plot→channel spectra and maps，it always reports errors: EEGLAB error in function welch() at line 87:
Incorrect use of zeros
The value of Size must be an integer

The file has been uploaded to Google drive, as shown below
https://drive.google.com/drive/folders/1QnhZySVRuG6oAD3vzw3zyvZ8VMQP5kbI?usp=sharing

The EEG data *.set was converted from *.edf files.

[image]

Please help me solve this issue.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*2020a*]  |
| EEGLAB version | [*2022.1*] |

AminAlam, 2023-10-21:
Hi,

I used the most recent dev version of EEGLAB with you datasets and it's working fine. Try upgrading to the most recent EEGLAB

a1phaeeg, 2023-10-23:
> Hi,
> 
> I used the most recent dev version of EEGLAB with you datasets and it's working fine. Try upgrading to the most recent EEGLAB

I'll try it, thank you!

arnodelorme, 2023-10-24:
Might be a problem with the fact the statistics toolbox is absent. I was able to load the file 1.set and plot the spectrum.

[image]

closed 2023-10-24 by arnodelorme

a1phaeeg, 2023-10-26:
The most recent dev version of EEGLAB  is working fine, thank you very much!