sccn/eeglab #691: I have an issue wheI have the same problem，matlab2019b eeglab2023.1

blink12138, 2023-10-22:
I have an issue when opening the manage extension from the EEGLAB (v2019.0) dropdown menu (File > Manage EEGLAB extensions) and it's throwing an error:

[image]

I've tried updating java, disabling the firewall and I have active internet connection yet the error still shows.

I need to download the LSL plugin.

_Originally posted by @arayes2000 in https://github.com/sccn/eeglab/issues/20#issuecomment-1594428833_

arnodelorme, 2023-10-24:
You can download plugins manually from this page

https://sccn.ucsd.edu/eeglab/plugin_uploader/plugin_list_all.php

closed 2023-10-24 by arnodelorme

blink12138, 2023-10-24:
[image]
[image]
I have tried adding it manually, but there seems to be a problem with the expansion of eeglab and it cannot be installed

I have tried:

1. Replace matlab2020b with matlab2019b

2. Change the version of eeglab from January 2019 to January 2023

3. Replaced Java version 1.8.202 to the latest Java11

4. Closed the firewall, used administrator running and compatibility mode

None of the above is useful

Is it related to updating the Windows 11 system or the files on the C drive

blink12138, 2023-10-24:
[image]
[image]
[image]

blink12138, 2023-10-24:
[image]
I have tried adding it manually, but there seems to be an issue with the expansion of eeglab and it does not display

blink12138, 2023-10-24:
Thank you very much for your reply and help. Thank you very much for any inconvenience caused

> 您可以从此页面手动下载插件
> 
> https://sccn.ucsd.edu/eeglab/plugin_uploader/plugin_list_all.php

Thank you very much for your reply and help. Thank you very much for any inconvenience caused

arnodelorme, 2023-10-24:
Yes, it is usually a firewall issue.

blink12138, 2023-10-24:
> 是的，这通常是防火墙问题。

But I completely closed my firewall and it was useless，Is there any other possible reason

blink12138, 2023-10-28:
[image]
[image]
I have found a solution, which is to establish a network proxy and unify the network proxy address of MATLAB with the proxy address of the computer

My computer system: windows11

MATLAB: matlab2020b

Eeglab: eeglab2023-1