sccn/eeglab #692: Adjust the color map on ERSP statistics plot

vvitkova, 2023-10-23:
Hello Eeglab community,

I have a study with 10 subjects and 2 experimental conditions, created in eeglab 2023.0, Matlab R2023a, Windows 11.

I am plotting ERSP spectrograms with permutations and Bonferroni corrections.
When I plot a spectrogram without specifying the power limits, I get an image which nicely shows the statistical differences, but is too saturated to show anything on the spectrograms.
[image]

When I plot the exact same graph by adding power limits, I get a better visualisation of the spectrograms but because of the color map, the statistical results become almost invisible (although they are there, they are a very, very pale yellow on a ligt blue background).
[image]

In the Edit tab -> Figure Properties, I tried using different color palettes. Some of them allow to better see the statistical differences but I am not happy with the results on the spectrogram side ("Prism" for example is my favorite "non-option" option, it gives quite psychadelic results :D ).
[image]

I am sure that during post-processing I can make the statistical results more visible but I would like to be able to see the differences easily during data exploration. Would there be a way to keep the power limits set in a way that it allows me to get an interpretable image on the spectrogram side, while adjusting the colop map of the statistical part so that the signifincat spots are more visible ?

Thank you very much in advance for your time and your ideas !

Best regards,
V.

arnodelorme, 2023-10-24:
The power limits should not affect significance. You can manually set the limits from the GUI (before calling the ERSP function).

closed 2023-10-24 by arnodelorme

vvitkova, 2023-10-24:
Dear Arnaud,

Thank you for your answer!
I am manually adjusting the power limits from the GUI. 
It is not the significance itself that is affected, it is the color map of the graphic that changes. 

When I adjust the power limits in a way to see the ERSP spectrograms well, the colors in the statistics map are also affected and the significance marks become very difficult to see. But they are not changed, the statistics results are the same as when I leave the power limits field empty in the GUI.

My question is how can I keep the colors in the statistics results more visible after adjusting the power limits in the GUI.

Best regards,
V.