sccn/eeglab #694: execution of script pop_loadbv as a function is not supported MAC IOS

betulyilmaz34, 2023-10-26:
### I am trying to load  a vhdr.file using EEGLAB.2021 with Mac. Such error attached keeps popping up. I also tried EEGLAB.2023 but still does not work. I do not how to fix it, İ'd really appreciate your help.

([image])))

<br>
#### Steps to Reproduce

1. File-> import data
2. Using EEGLAB functions and plugins -> from brain vision recorder vhdr or hdr
3. error (seen in the screenshot)

<br>
#### Expected behavior:
[import data and open the file in eeglab]

<br>
#### Actual behavior:
[*execution of script pop_loadbv as a function is not supported*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*12.2.1*] |
| Matlab version | [*R2023b*] |
| EEGLAB version | [*v2021.1*] |

AminAlam, 2023-11-10:
Hi, can you please share the data file? This way we can better check the source of error

betulyilmaz34, 2023-11-30:
Sure, I’ve attached a file I am trying to analyze, which is a vhdr file.
Thank you,
Betül 

￼

closed 2023-11-30 by arnodelorme