sccn/eeglab #701: instability in 2023.1 when loading studies

alenarto, 2023-11-09:
### Description

We're having issues of instability with eeglab 2023.1 that are difficult to pinpoint. Yesterday we ran into issues loading a study, with errors never seen before such as about having issues with function pop_loadset.m, e.g., a Matlab internal error saying that there's a function lacking the "end" statement.

I couldn't localize the error (as we've never modified these functions and they worked before), so I completely re-installed eeglab2023.1. 

It worked for a moment (the previously unloadable study was now loaded), and then upon a future loading of the study it started spitting out errors such as in the error link. Error 2 ("Reference to non-existent field 'novodatchans'" in insertchans() at line 321) was reproduced by two people working on two different studies. 

[Errors Link](https://drive.google.com/drive/folders/1hqpmEcSIiKTxuRYqztamwGOPUopDsvPJ?usp=sharing)

<br>
#### Steps to Reproduce

1. I am sorry but this is difficult to reproduce. For us it seems to require loading the study, any study. I've added a small study into the drive at link above as a sample of a study that generated error2 on the last attempt (but it also did generators the other errors, but didn't generate them upon reinstall of 2023.1).

<br>
#### Expected behavior:
A study to load.

<br>
#### Actual behavior:
Variable unsystematic errors are produced as above. Seems to happen after several users use the same version of this eeglab. Ok for a little while after a reinstall.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Centos7 on HCP cluster] |
| Matlab version | [9.8.0.1396136 (R2020a) Update 3] |
| EEGLAB version | [2023.1] |

arnodelorme, 2023-11-09:
I believe the first 2 errors are due to a typo (inserted by you :-) in eeg_checkchanlocs.m because I have checked the code and there is no "novodatchans" variable in the code. 

For the last error, I have added another check to the code. Let me know.

closed 2023-11-09 by arnodelorme

alenarto, 2023-11-09:
Hi Arno. For "novodatchans" you're right - the corrupt version contains a typo that someone inadvertently inserted. The other errors are a little trickier but we've reinstalled and will report if we encountered them again. THanks!

arnodelorme, 2023-11-09:
OK, thanks. Let me know