sccn/eeglab #702: 调用“readbvcon"时，未对输出参数“CONF”(可能还包括其他参数)赋值。

zxsjcf, 2023-11-18:
[image]
[image]
[image]
[image]
I am a novice, and eeglab encountered a problem when importing vhdr files. The conf function is using the latest version of eeglab. How can I solve this problem

Allexxann, 2023-12-31:
Looks like a junk file from MacOSX. If so, it cannot be opened, as it is not a real datafile.

arnodelorme, 2024-02-09:
Correct. This is not a valid .vhdr file. There should also be a .vmrk and .eeg/.dat file. Closing this for now.

closed 2024-02-09 by arnodelorme