sccn/eeglab #706: EEGLAB error in function plugin_menu()

DanielaRazo, 2023-11-30:
### Description

[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]

<br>
[image]

#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
Unrecognized field name "downloads"

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |

AminAlam, 2023-11-30:
Are you trying to install a new plugin? Please indicate the version of EEGLABas well.

DanielaRazo, 2023-11-30:
> Are you trying to install a new plugin? Please indicate the version of EEGLABas well.

Yes, EEGLAB loads normally but when I want to install a new plugin that appears. I have the latest version, 2023.1

AminAlam, 2023-12-01:
I've tried this and it works with Matlab 2023b and EEGLAB 2023. Based on how plugin struct is made [here](url), this error shouldn't be raised. Can you please mention the exacts steps which result in this error?

arnodelorme, 2024-02-09:
I am closing this for now. Let us know if the error can be reproduced and how.

closed 2024-02-09 by arnodelorme