sccn/eeglab #707: Error in EEGLAB plug in BrainBeats for HEP extraction

oux20a, 2023-12-06:
### Description
I am using the BrainBeats plug in feature of EEGLAB, hoping to use this for extracting and analysing Heartbeat Evoked Potentials from my EEG data.  I've solved a lot of previous error messages by installing plug ins, but I am new to EEGLAB and at a loss how to fix this one.  Any help appreciated. It's quite a new plugin, so I'm using the only version (BrainBeats1.1) which can be downloaded here: [https://sccn.ucsd.edu/eeglab/plugin_uploader/plugin_list_all.php](url)

I have put my biosemi .bdf EEG data into EEGLab on Matlab and saved as a .set file after assigning channel locations and rereferencing to mastoids.  I have then opened .set file in Brainbeats1.1 and tried to extract Heartbeat Evoked Potentials. 

Brainbeats paper contains instructions 
Cannard, C., Wahbeh, H., & Delorme, A. (2023). BrainBeats : an open-source EEGLAB plugin to jointly analyze EEG and cardiovascular ( ECG / PPG ) signals. 0–26.  
[https://www.biorxiv.org/content/10.1101/2023.06.01.543272v1](url)

I can run the script (method 1) until receiving error message.
[https://bangoroffice365-my.sharepoint.com/:i:/g/personal/oux20a_bangor_ac_uk/EVe4OLSVIyxOuWP8BexeLNwBUFrzIHEv5-QEt_JUzVkW1A?e=kpTywH]url

[image]
](url)

Here is the .set data file I am working with
[https://bangoroffice365-my.sharepoint.com/:u:/g/personal/oux20a_bangor_ac_uk/EVRMkBXb48JNhWAS7lfg2uYBj089XIXJr2SCrgRy3tvM5Q?e=bcw7M1](url)

<br>
#### Steps to Reproduce

1. [*Saved my .bdf file as a .set file in EEGLAB*]
2. [*Load data into EEGLAB: File > Load existing dataset >MGpractice231123_1524.set*]
3. [*Tools>Brainbeats>process file at subject level*]
4. [*Select "Heartbeat-evoked potentials HEP" as analysis to run, "ECG" as heart data type,  select the ECG channels labeled > 'EKGl' is my heart beat channel*]
5. [*All other ticks in boxes are left as automated*]
6. [*Select Ok*]

<br>
#### Expected behavior:
[*The paper  by Cannard et al 2023 doesn't specify what happens next - expected all the script to run on MATLAB and then produce some HEP data to analyse*]

<br>
#### Actual behavior:
[*I get an EEGLAB error message stating: "EEGLAB error in function run_HEP() at line 41: Assigning to 2 elements using a simple assignment statement is not supported. Consider using comma-separated list assignment.*]

<br>
#### Versions
|   |   |
|----| ---- | 
| OS version     | [*Windows 10 OS build 19045.3693*] |
| Matlab version | [*R2023b*] |
| EEGLAB version | [*2023.1*] |

arnodelorme, 2024-02-08:
Apologies for the late reply.
@amisepa can you look into this?

amisepa, 2024-02-08:
Can you please try the new version of BrainBeats v1.4. and see if the problem is still present? 
Let me know if it works. I can't access your dataset so please re-upload if you want me to take a look. 
Thanks

oux20a, 2024-02-09:
Hello Cedric

Thank you so much for getting back to me.  I am a PhD student doing research on interoception and was absolutely delighted to read your paper on BrainBeats last year. It would be wonderful to get it working for my research!

Things have moved on since my original post on Github.  Some error messages were resolved by downloading additional EEGLAB plugins.  I had downloaded the latest version, and this week also downloaded the individual functions that have been updated since the release of version 1.4.   However, it still only runs so far before error messages.   This is the latest error message:
"EEGLAB error in function ref_infinity() at line 152: Unable to perform assignment because the size of the left side is 1-by-1 and the size of the right side is 0-by-0.
My data set is attached, however I also get error messages using the BrainBeats tutorial data set.

If you can be of any help at all it would be very much appreciated.  I've asked an IT technician at my University for help but he is also struggling to get beyond error messages.

Thank you and best wishes
Kitty Forster

[https://res.cdn.office.net/assets/mail/file-icon/png/generic_16x16.png]MGpractice231123_1524.set<https://bangoroffice365-my.sharepoint.com/:u:/g/personal/oux20a_bangor_ac_uk/Ec6TdSBBb7RBpbGycbR4N6sBDtl_TBK1kfLdogxy0vyvDA>

Mae croeso i chi gysylltu gyda'r Brifysgol yn Gymraeg neu Saesneg. Ni fydd gohebu yn Gymraeg yn arwain at oedi.

You are welcome to contact the University in Welsh or English. Corresponding in Welsh will not lead to delay.

Rhif Elusen Gofrestredig 1141565 - Registered Charity No. 1141565

Gall y neges e-bost hon, ac unrhyw atodiadau a anfonwyd gyda hi, gynnwys deunydd cyfrinachol ac wedi eu bwriadu i'w defnyddio'n unig gan y sawl y cawsant eu cyfeirio ato (atynt). Os ydych wedi derbyn y neges e-bost hon trwy gamgymeriad, rhowch wybod i'r anfonwr ar unwaith a dilewch y neges. Os na fwriadwyd anfon y neges atoch chi, rhaid i chi beidio a defnyddio, cadw neu ddatgelu unrhyw wybodaeth a gynhwysir ynddi. Mae unrhyw farn neu safbwynt yn eiddo i'r sawl a'i hanfonodd yn unig ac nid yw o anghenraid yn cynrychioli barn Prifysgol Bangor. Nid yw Prifysgol Bangor yn gwarantu bod y neges e-bost hon neu unrhyw atodiadau yn rhydd rhag firysau neu 100% yn ddiogel. Oni bai fod hyn wedi ei ddatgan yn uniongyrchol yn nhestun yr e-bost, nid bwriad y neges e-bost hon yw ffurfio contract rhwymol - mae rhestr o lofnodwyr awdurdodedig ar gael o Swyddfa Cyllid Prifysgol Bangor.

This email and any attachments may contain confidential material and is solely for the use of the intended recipient(s). If you have received this email in error, please notify the sender immediately and delete this email. If you are not the intended recipient(s), you must not use, retain or disclose any information contained in this email. Any views or opinions are solely those of the sender and do not necessarily represent those of Bangor University. Bangor University does not guarantee that this email or any attachments are free from viruses or 100% secure. Unless expressly stated in the body of the text of the email, this email is not intended to form a binding contract - a list of authorised signatories is available from the Bangor University Finance Office.

oux20a, 2024-02-13:
Hello Cedric,

Thank you so much for offering your assistance.  I was confident I had the newest versions, but I’ve still wiped everything and followed the steps you outlined above.  

At first I got an EEGLAB error “EEGLAB error in function clean_eeg() at line 143: Undefined function ‘ref_infinity’ for input arguments of type ‘struct’.” 

This was a familiar error message that had previously been resolved following advice from a technician at my University - that I install the 'REST_cmd' plugin. 

I installed reference_infinity-REST_cmd again for my new EEGLAB download.  I also re-added plug in biosig4octmat-3.8.3 as this was another recommendation I’d found online previously for processing EEG/biomedical data.  My Matlab command window stated I wasn’t using the most updated version of plug-ins ‘dipfit’ and ‘ICLabel’ so I updated these too.  All plug-ins were sourced directly from https://sccn.ucsd.edu/eeglab/plugin_uploader/plugin_list_all.php  

It successfully runs through a lot more code, before the error: “EEGLAB error in function vis_artifacts() at line 153: Undefined function ‘mad’ for input arguments of type ‘double’”  

I’d read that EEGLAB may function better with additional toolboxes in Matlab (I am running most recent version R2023b) https://sccn.ucsd.edu/eeglab/ressources.php#:~:text=Additional%20Matlab%20toolboxes%3A%20EEGLAB%20requires,However%2C%20some%20toolboxes%20are%20recommended … So I’ve added Signal Processing toolbox, Statistics toolbox and Image processing toolbox.  

Running Brainbeats again, it went much further than I had seen previously (pop-up graphs included RR Intervals and artifacts, EEG channels removed, Interbeat intervals (IBI) distribution, Epochs removed). However, I got this error message  -first one I’ve seen that has said it’s not a bug: 

“pop_runica error: wrong value for argument ‘icatype’, This is not a bug (Error occurred in function pop_runica() at line 301)”

Any advice regarding how to move forward would be most appreciated.

Best wishes,

Kitty

amisepa, 2024-02-13:
Thanks for this new info. 

What computer are you using (Windows, MAc, Linux)?

Strangely, the REST_cmd plugin should get installed automatically by BrainBeats if it is not already installed. 
For future reference, you can also install and update plugins from the EEGLAB GUI, it's easier. File > Manage extensions > type name of plugin in search bar > Install/update

Strange about the error in vis_artifacts(), the mad function was introduced in MATLAB R2006a. Maybe part of the Statistics toolbox? not sure I thought that was part of basic MATLAB. 

“pop_runica error: wrong value for argument ‘icatype’, This is not a bug (Error occurred in function pop_runica() at line 301)”
Can you put a break point at line 301 of pop_runica and see what the input right after 'icatype' is. My guess is that the PICARD plugin was not installed automatically just like REST_cmd, and it's trying to run it. 
If you don't know how to do that, go extension manager and install the PICARD plugin manually. 
Alternatively, Change the option in the GUI parameters to the standard ICA and tell me if that works. 

Cedric

oux20a, 2024-02-14:
Hello Cedric,

I'm using Windows 11, it's a brand new laptop with 16GB RAM.  
Thanks for the tip re installing future plugins.  
Perhaps because my Matlab download was through a student license it was basic, but I could get add-on toolboxes.

After installing PICARD plugin it runs for longer again and the “pop_runica error” doesn’t happen.  I re-read your paper yesterday and saw reference to PICARD plugin there too, apologies for not picking up on this earlier.  

I had a go at the break point recommendation (I’m not a programmer as you’ve probably guessed!).  I put a break point at line 301 in pop_runica. When I ran Brainbeats again it stopped at that line and last two lines in command window read: 
“Pop_select: removing 18 unreferenced events
301 if ischar(g), error(g); end”

On the pop_runica code itself, I could only see “icatype” nearby at line 293. Around that the code reads:
[ g, addoptions ] = finputcheck( options, { 'icatype'        'string'  unique({ allalgs.name })   'runica'; ...
                            'dataset'        'integer' []        [1:length(ALLEEG)];
                            'options'        'cell'    []        {};
                            'concatenate'    'string'  { 'on','off' }   'off';
                            'concatcond'     'string'  { 'on','off' }   'off';
                            'reorder'        'string'  { 'on','off' }   'on';
                            'chanind'        { 'cell','integer' } { [] [] }        [];}, ...
                            'pop_runica', 'ignore');

That’s probably not what you meant and I've misunderstood? (But it runs further with PICARD anyway, which has resolved the pop_runica error).  

I tried running again but setting the "ICA method to extract artifactual components" to 'normal' rather than 'fast (default)' in the GUI for HEP.  It’s the only parameter I could see relating to your suggestion “Change the option in the GUI parameters to the standard ICA” - so I’m not sure if that was what you meant?  

I get “uicontrol error. Value must be numeric and finite, This is not a bug (Error occurred in function pop_selectcomps() at line 216).”  

I get this same error if I re-select the default ICA parameter of 'fast' as well.

Sorry this isn't an easy  fix, I'm very grateful for your support.

Thanks, 
Kitty

amisepa, 2024-02-15:
The plugin should be installed automatically by BrainBeats though. I'll need to investigate that. 

> "ICA method to extract artifactual components" to 'normal' rather than 'fast (default)'

Yes that's what I meant, sorry if that wasn't clear. 

> I get “uicontrol error. Value must be numeric and finite, This is not a bug (Error occurred in function pop_selectcomps() at line 216).”

Does this happen when you click in the GUI or while it runs? 

> After installing PICARD plugin it runs for longer again and the “pop_runica error” doesn’t happen. I re-read your paper yesterday and saw reference to PICARD plugin there too, apologies for not picking up on this earlier.

So this worked then? Is it only when you select new options in the GUI that you get an error now?

oux20a, 2024-02-15:
Hello Cedric,

The pop_runica error was resolved by PICARD plugin. 

The 'uicontrol error' is the next block - after PICARD plugin fixed pop_runICA I tried running again and got that error message (yes, with default parameters re-set in GUI).  

Sorry about this 

Thanks, Kitty

oux20a, 2024-02-28:
Hello Cedric,

I'm sorry to keep bothering you about this and really appreciate all your suggestions so far.  

I saw there had been updates to BrainBeats 2 days ago by yourself on https://github.com/amisepa/BrainBeats/ but unfortunately the error I described previously is still occurring, using the Brainbeats sample data.  
“uicontrol error. Value must be numeric and finite, This is not a bug (Error occurred in function pop_selectcomps() at line 216).”

I've been busy with the adventure therapy side of my interoception research over the last couple of weeks and am now focusing on progressing with the EEG data.   I'm going to look into alternatives for HEP but would ideally really love to use Brainbeats, so it would be great to hear if you think the error is resolvable.  

Best wishes, 
Kitty

amisepa, 2024-02-28:
Hi, yes sorry I have been working on other things, the paper is about to get published after a few minor plot tweaks. 

I thought you could run it without the GUI, no? Do you know how to use the command lines? 
See the "brainbeats_tutorial" script in the BrainBeats folder. 
For example, with all default settings, use this for HEP:

EEG = brainbeats_process(EEG,'analysis','hep','heart_signal','ECG', ...
    'heart_channels',{'ECG'},'clean_eeg',true);

amisepa, 2024-02-28:
I'll try to fix the GUI issue shortly

oux20a, 2024-02-28:
Ah sorry, miscommunication.  I meant 'yes' I still get the uicontrol error even with default parameters on GUI.  
Absolutely no need to apologise, I appreciate you will be very busy.  Please don't feel obliged to reply until the publication pressure is over.
If I run by command line I still get the same issue I think. It runs for the same amount of time (same graphs and text running in command window) then stops due to an error.  Rather than pop up box the error is specified in command window in red text. However it gives more detail afterwards than the GUI does, maybe that's helpful?  

 It reads: 
"Error using uicontrol
Value must be numeric and finite

Error in pop_selectcomps (line 216)
	hh = uicontrol(gcf, 'Style', 'pushbutton', 'string', 'Cancel', 'Units','Normalized', 'backgroundcolor',
    GUIBUTTONCOLOR, ...

Error in clean_eeg (line 304)
                pop_selectcomps(EEG,1:nComps)

Error in run_HEP (line 110)
    [HEP, params] = clean_eeg(HEP,params);

Error in brainbeats_process (line 290)
        EEG = run_HEP(EEG, CARDIO, params, Rpeaks); % for adding cardio channel back in final output

Error in brainbeats_tutorial (line 71)
EEG = brainbeats_process(EEG,'analysis','hep','heart_signal','ECG', ..."

oux20a, 2024-02-28:
And the point that it gets to before the error is:
_[lots of previous iterations...]_
iteration 172, gradient norm = 1.602e-08
Scaling components to RMS microvolt
Scaling components to RMS microvolt
ICLabel: extracting features...
Scaling components to RMS microvolt
Recomputing ICA activations
ICLabel: calculating labels...
ICLabel: saving results...
0 components flagged for rejection, to reject them use Tools > Remove components from data
Scaling components to RMS microvolt
Drawing figure...
**Error using uicontrol**

amisepa, 2024-02-28:
Oh ok, so this is actually an ICLabel issue, when plotting the components. 

@arnodelorme  have you ever seen this when calling pop_selectcomps ?

arnodelorme, 2025-02-07:
I have not seen that issue before. Try to give me a dataset to reproduce running ICLabel and creating the plotting error.

arnodelorme, 2025-09-13:
I cannot reproduce the problem. It must be a platform specific problem.

closed 2025-09-13 by arnodelorme