sccn/eeglab #710: EELAB error in importing .edf file from Enobio (EEG) system

Budha-Kh-NRC, 2023-12-14:
### Description

[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]

 Hi,

When we import .edf file (from Enobio, EEG system) with triggers for stimuli and response corresponding to a 5-minute GO/NO-GO task into EEGLAB using Biosig, we run into an error - 

**EEGLAB error in function edfannot2evt() at line 57:

Index exceeds the number of array elements (1).

If you think this is a bug, please submit a detailed description of how to reproduce the problem (and upload a small dataset).**

However for shorter sessions (1 min), we did not have the error and the .edf file is imported. Can anyone help in fixing this?

We also installed import_edf and tried importing the data with it, but got this error

**File size in bytes is 2680052. File size must be 2658828 as specified in 

This is not a bug (Error occurred in function validateEDF() at line 57)**

Here is the data
https://drive.google.com/file/d/1TYhwxHCvZrMMFxVUtWAO1UkxDE7Whh73/view?usp=drive_link

Thank you in advance,

Budha

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Windows*] |
| Matlab version | [*2023b*] |
| EEGLAB version | [*2023.1*] |

Teuniz, 2023-12-16:
That file is severely broken. It has at least two errors and the scaling/sensitivity of the signals is weird.
Which software produced that file?

Budha-Kh-NRC, 2024-01-08:
> ***Attention*** This email originated from outside of the NRC. ***Attention*** Ce courriel provient de l'extérieur du CNRC.

Hi,

Thank you for your response.

This .edf file was generated by NIC software of the Enobio EEG system. I am attaching here another .edf file, would you try with this one and see if this one is also broken as I still get the same error while importing to EEGlab.

Thank you in advance,

Best,

Budha

arnodelorme, 2024-01-08:
@amisepa can you have a look at this EDF file?

amisepa, 2024-01-08:
Couldn't import using Matlab's edfread function either. 

There is a discrepancy between the actual file size and the expected size stored in the file header. I think this file is corrupted. 

[image]

Cedric

amisepa, 2024-01-08:
Confirmed by the EDFbrowser software as well:
[image]

amisepa, 2024-01-08:
I cannot easily find how to adjust the info in the header as they suggest. 

See also this post on the same topic, maybe useful: https://github.com/holgern/pyedflib/issues/58

closed 2024-01-08 by amisepa

arnodelorme, 2024-01-08:
Thanks @amisepa

Budha-Kh-NRC, 2024-01-08:
Thank you Cedric and Arnaud. I will follow up with folks at Enobio regarding the error in file size.

Best,

Budha

Thanks @amisepa

Budha-Kh-NRC, 2024-01-23:
Hi Cedric,

Thank you once again for the reply.

I am in touch with Enobio to figure out this discrepancy. One thing though, when we record the task EEG data for 1 minute only (instead of 4 mins), we are able to import the .edf into EEGLAB. Would you have a look in this smaller .edf file and see if there is still discrepancy?

Best,

Budha

image.png (view on web)<https://github.com/sccn/eeglab/assets/58382227/ef28fa07-0bcd-4aa8-8beb-49f891fa640b>