sccn/eeglab #714: Could not find associated Limo file (LIMO EEG)

naoqo, 2023-12-23:
### Description
I have completed a first level (subject) analysis using EEGLAB plugin LIMO EEG.
But when I try to carry out 2nd level analysis, matlab returns following errors. 

This error happens after I selected the text file which contains the directries of beta files for each participants.
The beta file looks like successfuly created after the 1st level analysis.
The execution leaves empty folders with names of seleceted analysis at 2nd level.

I tried all the inference analysis listed in GUI but they all returns the same error message.
In my study design, the independent variable contains 2 categorical values (S-1, S-2, plus constant). 
The data include 2 groups of participants which differ in number.

I'm not sure if this is a bug or just an error due to my data.
But I'd greately appreciate your could provide any assistance with fixing this issue.

Thanks,
Naoko

[*Description of the bug*]
Warning: could not find associated LIMO file
  inputdlg2() - inputdlg function clone with coloring and help for 
                eeglab().
 
  Usage:
    >> Answer = inputdlg2(Prompt,Title,LineNo,DefAns,funcname);
  
  Inputs:
    Same as inputdlg. Using the optional additional funcname parameter 
    the function will create a help button. The help message will be
    displayed using the pophelp() function.
 
  Output:
    Same as inputdlg
 
  Note: The advantage of this function is that the color of the window
        can be changed and that it displays an help button. Edit 
        supergui to change window options. Also the parameter LineNo
        can only be one.
 
  Author: Arnaud Delorme, CNL / Salk Institute, La Jolla, 11 August 2002
 
  See also: supergui(), inputgui()

Output argument "result" (and possibly others) not assigned a value in the execution with "inputdlg2" function.

Error in limo_inputdlg (line 4)
    res = inputdlg2(varargin{:});

Error in limo_random_select>get_beta_indices (line 2296)
    param = eval( [ '[' cell2mat(limo_inputdlg('which parameters to test e.g [1:3]','parameters option')) ']' ]);

Error in limo_random_select>check_files (line 2359)
                [parameters,betas] = get_beta_indices(selectmode, fullfile(Paths{1}, Names{1}));

Error in limo_random_select (line 195)
        parameters = check_files(Paths, Names,1);

Error in limo_random_effect>One_Sample_t_test_Callback (line 194)
        limo_random_select('one sample t-test',handles.chan_file,'nboot',handles.b,'tfce',handles.tfce,'type','Channels');

Error in gui_mainfcn (line 95)
        feval(varargin{:});

Error in limo_random_effect (line 36)
    gui_mainfcn(gui_State, varargin{:});

Error in matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)limo_random_effect('One_Sample_t_test_Callback',hObject,eventdata,guidata(hObject))
 
Error while evaluating UIControl Callback.

<br>
#### Steps to Reproduce

I followed workflow recommended in LIMO eeg tutorials and paper which explains how to use limo from bids formated study data.

https://www.frontiersin.org/articles/10.3389/fnins.2020.610388/full

https://github.com/LIMO-EEG-Toolbox/limo_meeg/wiki/2.-One-way-repeated-measures-ANOVA-(Famous,-Unfamiliar,-Scrambled-faces-as-conditions

(After 1st level analysis)
1. Study --> Linear modeling of EEG data
2. 2nd level analysis (Opens the gui)
3. Select ANOVA/ANCOVA, paired t-test, etc
4. Select Beta file directly from pop-up menu
5. Error is returned

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Windows 11 pro*] |
| Matlab version | [*2023a*] |
| EEGLAB version | [*2022.0*] |

arnodelorme, 2024-02-08:
@CPernet any comment on this LIMO problem?

CPernet, 2024-02-08:
Hi Naoko, 

Which version of LIMO are you using? v3.4 is the stable version.

If you use the version that comes with EEGLAB plug-in manager (v4), it as been updated a lot as the new GUI generates lots of errors -- most are now fixed here: https://github.com/LIMO-EEG-Toolbox/limo_tools/tree/master. I believe I have seen that error before and fixed it. If that the case that you use the one from the plug-in manager (ie v4) please use the github version and let me know what is happening. If the error subsist, I'll fix it. 

thanks
cyril

closed 2025-02-07 by arnodelorme