sccn/eeglab #720: Cannot adjust event latencies

Loluer, 2024-01-22:
### Description

I am working with EGI.mff files and have imported event type fields, including the 'code' and 'mffkey'. I've encountered latency setting errors and need to adjust the event type latencies by -200ms.

However, I'm facing an issue in EEGLAB where, despite the event types being clearly displayed in both the event type options and the scrolled channel data, there seems to be a bug. Specifically, when trying to adjust latencies, EEGLAB reports that the requested event types are not found.

It's worth noting that I can successfully extract epochs using the same event types. I'm puzzled about why I cannot adjust latencies and would like guidance on how to debug this issue. Any insights or suggestions would be greatly appreciated.

<br>
#### Steps to Reproduce

1. import data
[image]

2. set the parameters to adjust latencies
[image]
[image]

3. reported errors from EEGLAB
[image]
[image]

4. succeed to extract the epochs using same event types
 EEG = pop_epoch( EEG, {  'CSTI_p1s3'  'CSTI_p2s5'  'CSTI_p3s1'  'CSTI_p4s2'  'CSTI_p5s4'  'ISTI_p1s1'  'ISTI_p1s2'  'ISTI_p1s4'  'ISTI_p1s5'  'ISTI_p2s1'  'ISTI_p2s2'  'ISTI_p2s3'  'ISTI_p2s4'  'ISTI_p3s2'  'ISTI_p3s3'  'ISTI_p3s4'  'ISTI_p3s5'  'ISTI_p4s1'  'ISTI_p4s3'  'ISTI_p4s4'  'ISTI_p4s5'  'ISTI_p5s1'  'ISTI_p5s2'  'ISTI_p5s3'  'ISTI_p5s5'  }, [-1  2], 'newname', 'zhuyuntian_formal epochs', 'epochinfo', 'yes');

5. event types
[image]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*windows 11*] |
| Matlab version | [*2021b*] |
| EEGLAB version | [*EEGLAB v2023.0*] |

arnodelorme, 2024-02-08:
Apologies for the late response. Cab you share a dataset so I can try. I cannot reproduce the problem.

arnodelorme, 2024-02-09:
I am closing this for now. Feel free to share a file and reopen the issue.

closed 2024-02-09 by arnodelorme