sccn/eeglab #721: Downloading EEG Lab Compiled to a Mac running Sonoma 14.2.1

SNAPLAB12, 2024-01-24:
### Description
I am not able to download the EEG Lab compiled software because my mac says it is not able to determine if it contains malware. I have tried several times and keep getting the same message. Is there a way around this?

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |

arnodelorme, 2024-01-24:
Yes, this is annoying. There is downloading and executing. This is the direct link https://sccn.ucsd.edu/eeglab/currentversion/eeglab_compiled_macos.zip I tried downloading it on my MAC and it was ok.

SNAPLAB12, 2024-01-24:
> You don't often get email from ***@***.*** Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>
> ***ATTENTION: This email came from an external source. Do not open attachments or click on links from unknown senders or unexpected emails.***

Good morning,

Thank you for the link. I tried it with the same result. Please find a screen shot of what happened attached.

I am not sure if there is something I need to change in settings. This has never happened to me before. I tried downloading with Brave, as well as Google Chrome to see if there would be a difference, but unfortunately to the same result both times.

Thanks!

arnodelorme, 2024-02-09:
There is no real way around that except to disable some security on your MAC.

closed 2024-02-09 by arnodelorme