sccn/eeglab #722: ICLabel error on Linux (default option)

tadeshi88, 2024-01-24:
ICLabel is not working on Linux with the default option, resulting in the below error.
 
[image]

The ICversions Lite/Beta functions seamlessly. All ICversions worked on two Windows machines I tested with identical Matlab/EEGLab versions. Thus, likely a Linux-specific issue.

The issue is within the following Matlab function:
 Matlab2023b/toolbox/signal/signal/+signal/+internal/+resample/uniformResample.m
An associated workspace has been saved to aid in debugging. The code execution is interrupted just before the line causing the crash.

[workspace.zip](https://github.com/sccn/eeglab/files/14037475/workspace.zip)

Versions

[image]
EEGLAB v 2023.1

arnodelorme, 2024-01-24:
We primarily use Linux (although not Ubuntu - CentOS) and have no problem with ICLabel. Can you share your dataset with ICA weights prior to running ICLabel?

arnodelorme, 2024-02-01:
@amisepa can you pitch in? Does this dataset has EEG?

amisepa, 2024-02-01:
We have already exchanged a few emails about this strange issue and I thought it might be a Linux issue since it works with on Windows and Mac. If this is the sample dataset from BrainBeats, it contains  64 EEG channels, 1 ECG channel and 1 PPG channel. BrainBeats automatically deals with the heart ones. Did you remove them here?

amisepa, 2024-02-01:
I loaded the file, there are only the 65 EEG channels, as expected. However, the ICA weights are not there, suggesting you did not run ICA (or at least you did not save the file after running it):
[image]

amisepa, 2024-02-01:
I ran ICA and ICLabel worked fine. Not sure what the issue could be.

arnodelorme, 2024-02-01:
Strange. Let’s close the issue and reopen it if we get more insight as to what the problem might be.