sccn/eeglab #725: pop_importepoch error

carrie372, 2024-02-07:
### Description

Trying to import epoch info following the instructions [linked](https://eeglab.org/tutorials/04_Import/Importing_Event_Epoch_Info.html#importing-epoch-info-matlab-array-or-text-file-into-eeglab) from EEGLAB tutorial. I get the error "There must be as many field names as there are columns in the file/array, This is not a bug (Error occurred in function pop_importepoch() at line 252)"

<br>
#### Steps to Reproduce

1. [load data set]
2. [import epoch]
3. [enter information as instructed]

sample data file for epoch to import
[image]

pop_importepoch window and what I input into the GUI
[image]

<br>
#### Expected behavior:
[epochs to be imported]

<br>
#### Actual behavior:
[error message described above]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [MacOS 10.15.7 and Windows 11 22621.3007] |
| Matlab version | [2023a and 2023b] |
| EEGLAB version | [2023.1] |

arnodelorme, 2024-02-08:
If you get an error that the number of trials do not match the number of trials in the EEG dataset, you might want to look at your dataset (how many epochs in this dataset) and the file you are trying to import (how many rows). I have checked that the tutorial data can be imported. I am closing this for now but feel free to reopen.

closed 2024-02-08 by arnodelorme

carrie372, 2024-02-08:
I have checked this and they are the same amount.

arnodelorme, 2024-02-08:
There should be one more row in the text file (because of the header). Is that the case?

carrie372, 2024-02-08:
yes