sccn/eeglab #726: Loading multiple existing datasets from GUI fails

RomainGrandchamp, 2024-02-08:
### Description

[*Error when trying to load multiple datasets selected from file selector GUI. https://drive.google.com/drive/folders/13B05CwhpeWV_XgGQRIMQMxtI12nSDiuE?usp=drive_link*]

<br>
#### Steps to Reproduce

1. Open menu File / Load existing dataset
2. Select multiple dataset files in the File selection window (pop_loadset())
3. Click "Open" button

<br>
#### Expected behavior:
EEGLAB should load all selected datasets

<br>
#### Actual behavior:
Dialog box opens with error message :  
"Too many input arguments. , 
This is not a bug (error occured in function pop_loadset() at line 240)"

Output in matlab : 
"pop_loadset(): loading file C:\data\ORPOSTEMO1_vection_S01_ST_EEG-averef.set ...
pop_loadset(): loading file C:\data\ORPOSTEMO1_vection_S01_V_EEG-averef.set ...
pop_loadset(): loading file C:\data\ORPOSTEMO1_vection_S02_ST_EEG-averef.set ...
pop_loadset(): loading file C:\data\ORPOSTEMO1_vection_S02_V_EEG-averef.set ...
pop_loadset(): loading file C:\data\ORPOSTEMO1_vection_S04_ST_EEG-averef.set ...
pop_loadset(): loading file C:\data\ORPOSTEMO1_vection_S04_V_EEG-averef.set ...
Unrecognized function or variable 'EEG'.
 
Error while evaluating Menu Callback."

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*MS Windows 10 Professional 64 bits Version 22H2 Build 19045.3930*] |
| Matlab version | [*R2020b*] |
| EEGLAB version | [*v2023.1*] |

LukeCepza, 2024-02-08:
For some reason it doesn't work when you try to do that in pop_loadset (from GUI). I think this is due to the GUI trying to load with the fullfile. To avoid the issue i load files from console. You can try like this: 

EEG = pop_loadset('filename','ORPOSTEMO1_vection_S01_ST_EEG-averef.set','filepath', 'C:\data\');
[ALLEEG, EEG, CURRENTSET] = eeg_store( ALLEEG, EEG, 0 )
EEG = pop_loadset('filename','ORPOSTEMO1_vection_S01_V_EEG-averef.set','filepath', 'C:\data\');
[ALLEEG, EEG, CURRENTSET] = eeg_store( ALLEEG, EEG, 0 )
EEG = pop_loadset('filename','ORPOSTEMO1_vection_S02_ST_EEG-averef.set','filepath', 'C:\data\');
[ALLEEG, EEG, CURRENTSET] = eeg_store( ALLEEG, EEG, 0 )

... and so on for all your files

What eeg_store() does is to save the current EEG in ALLEEG and asign a set number

Hope this solves the issue :)

RomainGrandchamp, 2024-02-08:
> https://github.com/sccn/eeglab/issues/726#issuecomment-1934790998 | view it on
> GitHub ] , or [

Hi Kevin, 
thanks for your reply. 
I am actually using scripting as you describe to load multiple datasets but I noticed this behavior by chance. 
It is actually working with the GUI in version v2023.0 so that's why I was wandering if anything changed and signaled it. 

Cheers 
Romain

arnodelorme, 2024-02-08:
I tried to reproduce but could not. All the files need to be in the same folder (which they are for you). Would you mind zipping the files and send them to me?

RomainGrandchamp, 2024-02-08:
Here they are : https://drive.google.com/file/d/1RfUcacLMqDRNWm149qVQifN5T7b73_Vd/view?usp=drive_link

arnodelorme, 2024-02-09:
Thanks @RomainGrandchamp 
I tried and it works for me. I can only assume that you are using a version of EEGLAB that is slightly older. If the problem occurs with the latest GITHUB version, please let me know.