sccn/eeglab #727: @arnodelorme when I follow the tutorial I end up with two Cz channels. One for channel 24 and the new one appended in channel locations.

carrie372, 2024-02-08:
@arnodelorme when I follow the tutorial I end up with two Cz channels. One for channel 24 and the new one appended in channel locations.

_Originally posted by @carrie372 in https://github.com/sccn/eeglab/issues/614#issuecomment-1932692502_

LukeCepza, 2024-02-08:
I think it is because Cz is an offline reference on your EEG system. Online references doesn't show up in eeg records because reference is already applied. Offline you would have to reference first to Cz, then Cz would dissapear from channel list. Afterwards (i guess) you would add again Cz and re-reference to scalp average. Please also share more details on your issue :)

carrie372, 2024-02-08:
"Offline you would have to reference first to Cz, then Cz would dissapear from channel list. Afterwards (i guess) you would add again Cz and re-reference to scalp average." What are the steps for this in EEGLAB? 

"Please also share more details on your issue" Can you please specify the details you would like? Below is a screenshot of vhdr file with Cz listed as reference. 
[image]

LukeCepza, 2024-02-08:
In the GUI you would do something like this

Specify reference Cz:
[image]

Compute scalp average reference and recover Cz reference.
[image]

However, I'm pretty sure it is equivalent as just doing scalp average reference. I think it would be better just to reference to Cz first, and then scalp average without trying to recover Cz as this helps to increase SNR.

Hope this helps :)

closed 2024-02-08 by carrie372