sccn/eeglab #728: How to separate GSR/ECG files saved together from Brain Vision and clean the ECG files

SNAPLAB12, 2024-02-13:
Hello, I have imported ECG files using the Brain Vision plug in, but each file also has GSR data that I do not need.

How do I separate the GSR and ECG data and save the ECG files so I can use Brain Beats to clean the ECG files?

arnodelorme, 2024-02-14:
@amisepa here is a BrainBeats question

amisepa, 2024-02-15:
Hi,

You can try to run BrainBeats directly, there is a feature to automatically detect and remove unnecessary auxiliary channels. However that is limited by how your GSR electrode is labeled. 

Try: 
EEG = brainbeats_process(EEG,'analysis','features','heart_signal','ECG', ...
    'heart_channels',{'ECG'},'eeg_features','off',...
    'hrv_features',{'time' 'frequency' 'nonlinear'},...
    'vis_cleaning',true,'vis_outputs',true);

If detected, you should see a pop-up window that asks you confirmation to remove it. 

Otherwise, do you use GUI or command line?

GUI: 
1) Edit > Select data > Click on the ... button next to Channel types > Select your GSR electrode in the list of labels > Check the box "Remove these" 
2) File > Save current dataset as > Selet location and save as .set file  (Or you can just run BrainBeats directly and it will save the file by default at the end)
3) Tools > BrainBeats > 1st level > Uncheck all EEG options you see

Command line: 
EEG = pop_select(EEG,'nochannel',{'your_GSR_electrode_label'});
EEG = brainbeats_process(EEG,'analysis','features','heart_signal','ECG', ...
    'heart_channels',{'ECG'},'eeg_features','off',...
    'hrv_features',{'time' 'frequency' 'nonlinear'},...
    'vis_cleaning',true,'vis_outputs',true);

SNAPLAB12, 2024-02-19:
I am using GUI, but when I click on tools, BrainBeats is not there. I am not sure how to add it. I am using EEG Lab Compiled on a PC. Do I need to download the updated version of EEG Lab to access Brain Beats on EEG Lab Compiled?

Thanks!

amisepa, 2024-02-20:
Oh sorry BrainBeats is not yet available from the compiled version. @arnodelorme Would BrainBeats work with Octave?

closed 2024-02-20 by amisepa