sccn/eeglab #730: ICLabel

helentobback, 2024-02-14:
To preprocess my EEG data I am using ICA in an automated way: see script below. This script was working great, but since a few days i get the next error(s):

Error using  / 
Matrix dimensions must agree.

Error in topoplotFast (line 894)
squeezefac = rmax/plotrad;

Error in ICL_feature_extractor (line 33)
            topoplotFast(EEG.icawinv(:, it), EEG.chanlocs(EEG.icachansind), ...

Error in iclabel (line 52)
features = ICL_feature_extractor(EEG, flag_autocorr);

Error in pop_iclabel (line 109)
    EEG = iclabel(EEG, icversion);

Error in test_ICA_2manyarguments (line 29)
        EEG = pop_iclabel(EEG, 'Default' );

I tried to manage this error by myself, but without succes. Do you have any idea of how to solve this?

Script:
%%Variable Declaration for the rest of the script
totalOfParticipants = 1;
totalOfSessions = 1;
path = X
outputPath = X
 
%%% Main preprocessing loop with P as a variable for the participant number and S for the
% sessions number
for P = 1:totalOfParticipants
    for S = 1:totalOfSessions
        
        % Generate file names
        inputFileName = X
        preProcessOutput = X
        FileNameBadChannels = X

        % Import data files into EEGLAB using BIOSIG toolbox. These data files include all channels, referenced to the mastoids (65-66) and including the EOGs (67-70). 
        EEG = pop_biosig(inputFileName, 'channels',1:70,'ref',[65 66] ,'refoptions',{'keepref','off'});
        
        EEG = pop_resample(EEG, 128);

        % Run ICA and flag artifactual components using IClabel
        EEG = pop_runica(EEG, 'icatype', 'runica', 'extended',1, 'chanind', 1:64);
        EEG = pop_iclabel(EEG, 'Default' );
        EEG = pop_icflag(EEG,[NaN NaN;0.8 1;0.8 1;NaN NaN;NaN NaN;NaN NaN;NaN NaN]);
        EEG = pop_subcomp(EEG, find(EEG.reject.gcompreject), 0);

    end
end

arnodelorme, 2024-02-14:
Would you mind sharing your dataset. I cannot reproduce the problem with the tutorial dataset. Thank you

arnodelorme, 2024-03-15:
Closing this for now. Feel free to reopen.

closed 2024-03-15 by arnodelorme