sccn/eeglab #732: Error using pop_eegstats; Too many output arguments.

stellafychang, 2024-02-21:
### Description

I was following [the eegstats tutorial](https://github.com/sccn/eegstats) but got an error after running the code below
```
[power, iafSum, iafChan, freqs] = pop_eegstats(EEG, 'thetarange',[4 6] ,'alpharange',[8 12] , ...
            'otherranges',[18 22;30 45],'averagepower','off','channels','','winsize',2, ...
            'overlap',1,'iaf','on','iafminchan',1,'alphaasymmetry','on','asymchans','F3 F4', ...
            'csvfile','save_these_results.csv');
```

<br>
#### Steps to Reproduce

1. type `eeglab cont` in the command window.
2. run the code below in the command window.
```
[power, iafSum, iafChan, freqs] = pop_eegstats(EEG, 'thetarange',[4 6] ,'alpharange',[8 12] , ...
            'otherranges',[18 22;30 45],'averagepower','off','channels','','winsize',2, ...
            'overlap',1,'iaf','on','iafminchan',1,'alphaasymmetry','on','asymchans','F3 F4', ...
            'csvfile','save_these_results.csv');
``` 

<br>
#### Expected behavior:<br>
I was expecting the same output as when I clicked GUI and an output file save_these_results.csv.
<br>
<br>
#### Actual behavior:<br>
I saw the error in the command window. <br>
Error using pop_eegstats <br>
Too many output arguments. <br>

However, I tried the code below and I did get 3 outputs from a, b, and c.
```
[a,b,c] = pop_eegstats(EEG, 'thetarange',[4 6] ,'alpharange',[8 12] , ...
            'otherranges',[18 22;30 45],'averagepower','off','channels','','winsize',2, ...
            'overlap',1,'iaf','on','iafminchan',1,'alphaasymmetry','on','asymchans','F3 F4', ...
            'csvfile','save_these_results.csv');
```

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*macOS Sonoma 14.3.1*] |
| Matlab version | [*R2023b*] |
| EEGLAB version | [*2024.0rc*] |

arnodelorme, 2024-02-23:
Thanks I have updated the header.

arnodelorme, 2024-02-23:
Note that the output section of the documentation was correct in case you are wondering what a and b are (c is the history).

closed 2024-02-23 by arnodelorme