sccn/eeglab #733: std_readdata.m setinds output error: Subscripted assignment between dissimilar structures.

stellafychang, 2024-02-21:
### Description

I was using the command line below in a STUDY and it was working until a specific dataset was included in the STUDY. This specific dataset was recorded with a different version of E-Prime than other datasets.
`[STUDY, datavals, xvals, yvals, events, params, setinds, datainfo] = std_readdata(STUDY, ALLEEG, 'channels', { ALLEEG(1).chanlocs.labels },'design',1)`

<br>
#### Steps to Reproduce
<br>
1. run std_precomp(STUDY, ALLEEG, { ALLEEG(1).chanlocs.labels }, 'erp', 'on','recompute','on');
<br>
2. run [STUDY, datavals, xvals, yvals, events, params, setinds, datainfo] = std_readdata(STUDY, ALLEEG, 'channels', { ALLEEG(1).chanlocs.labels },'design',1);

<br>
<br>
#### Expected behavior:
<br>
setinds field will have output which corresponds to dataset indices.

<br>
<br>
#### Actual behavior:
<br>
I got an error if this specific data is included into the STUDY.

> Subscripted assignment between dissimilar structures.
> 
> Error in std_readdata>reorganizedatastruct (line 456)
>                 datavals{iItem}(count:count+numItems-1) = dataTmp{iCase}{iItem};
> 
> Error in std_readdata (line 390)
>     datainfo = reorganizedatastruct(dataTmpSubj);
>  
If this specific dataset is removed from the STUDY, the setinds field will have output.
<br>
#### Question:
Is there any way to include this specific dataset in STUDY if we would like to have setinds field output? It would be fine if this dataset needs to be excluded.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*macOS Sonoma 14.3.1*] |
| Matlab version | [*R2023b*] |
| EEGLAB version | [*2024.0rc*] |

arnodelorme, 2024-02-23:
This is puzzling indeed. Can you share your data?

closed 2024-02-23 by arnodelorme

stellafychang, 2024-02-24:
It works on my end also. Thank you.