sccn/eeglab #738: Importing .EEG

Faqani1993, 2024-02-28:
### Description

I wanted to import my egg file from **Neuroscan .EEG file** plugin, but I received the error. 

EEG lab error  in function loadeeg() at line 253.
Unrecognised function or variable 'chann_names'.

[image]

| OS version     | [MacBook Pro 2023 version 13.5 |
| Matlab version | [R2023b] |
| EEGLAB version | [v2023.1] |

arnodelorme, 2024-02-28:
Are you sure it is a Neuroscan file? Can you share the file?

Faqani1993, 2024-02-28:
> Are you sure it is a Neuroscan file? Can you share the file?

My data is in EEG format. 

https://1drv.ms/u/s!AmRplatHNX8NaycgjGdjBKpDcLg?e=a1nftH

Thank you.

Faqani1993, 2024-02-29:
> Are you sure it is a Neuroscan file? Can you share the file?

I recorded the EEG signals with an MITSAR amplifier and WINEEG software.

arnodelorme, 2024-02-29:
Yes, this is not neuroscan. You need to export to EDF from WINEEG (reload the .eeg file in WINEEG and export it as EDF). Best

Faqani1993, 2024-02-29:
Thank you so much.