sccn/eeglab #739: error in function std_precomp()

ydf-ysq, 2024-03-04:
### Description
Hello, 
I was trying to precompute channel measures in my STUDY. I thought it was a data problem at first since EEGLAB worked well last year, so I tried this again in animal_study provided by the official website, but it failed again. I try to precompute all the measures  but only ERP-image worked, and all the failed ones pop up a window showing "can't find the source code for the parfor loop that attempts to execute the worker process".  Has anyone experienced a similar problem? Or does anyone know what might lead to this problem? 
[image]
[image]
[image]

<br>
#### Steps to Reproduce

1. Load existing study-> animal.study
2. Study->Precompute channel measures
3. Check "ERPs" or "Power spectrum", and click "OK"
4. Error appeared

<br>
#### Expected behavior: Computation of channel measures, so I can  plot them.

<br>
#### Actual behavior: poping up a window showing "can't find the source code for the parfor loop that attempts to execute the worker process" which I described above.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | Windows 10 64 bit |
| Matlab version | R2023a|
| EEGLAB version | 2023.0|

arnodelorme, 2024-03-05:
Which Animal study did you use. The one on that page? https://eeglab.org/tutorials/tutorial_data.html

I have just re-downloaded the data and tried and could not reproduce the problem.

ydf-ysq, 2024-03-05:
> Which Animal study did you use. The one on that page? https://eeglab.org/tutorials/tutorial_data.html
> 
> I have just re-downloaded the data and tried and could not reproduce the problem.

I downloaded it from this page https://eeglab.org/tutorials/10_Group_analysis/study_data_visualization_tools.html
[image]

arnodelorme, 2024-03-05:
Yes, it is the same data. Can you share the exact error message?

ydf-ysq, 2024-03-05:
> Yes, it is the same data. Can you share the exact error message?

The two lines in the middle mean "incorrect use of std_precomp" and "can't find the source code for the parfor loop that attempts to execute the worker process", respectively.
Is there any other information needed?
[image]

ydf-ysq, 2024-03-05:
For some reason, I installed E-prime2.0 last month, and I also installed .net3.5 runtime library because running E-prime2.0 required it. I uninstalled E-prime2.0 but it still failed to work. Then I tried to uninstalled .net3.5 which I was not sure if it was successful, but EEGLAB still produce the problem.

ydf-ysq, 2024-03-10:
> Yes, it is the same data. Can you share the exact error message?

Hello, I have reinstalled Matlab and EEGLAB in the past few days, but it still failed to work. I also commented on the code prompted by error message when precomputing ERP, trying to find why the error occurred. Then I found that when I commented on the code like this, the error would be reported. I have no idea about how to solve this problem, can you give me some advice so that I can try to solve it myself? 
Thank you very much!
[image]

arnodelorme, 2024-03-10:
Which version of EEGLAB are you using?

ydf-ysq, 2024-03-11:
> Which version of EEGLAB are you using?

2023.1

arnodelorme, 2024-03-11:
Can you try with 2024.0?

ydf-ysq, 2024-03-11:
> Can you try with 2024.0?
> […](#)

Sorry, I'm not very familiar with Matlab. I just found that I can switch the language to English. And below is the error message when I used 2023.1 to precompute ERPs.
[image]

And I also try with 2024.0, but I tried several times and I could not get it started.
[image]

ydf-ysq, 2024-03-12:
> Can you try with 2024.0?
> […](#)

Hello, I have solved the problem. I found that this problem would appear if I put the eeglab folder into the toolbox folder of Matlab. 
Thank you very much for taking time to reply me and help me solve the problem.

closed 2024-03-12 by arnodelorme