sccn/eeglab #741: EEGLAB error in function pop_micro_selectdata() at line 172:

Githubwl66, 2024-03-04:
I am having the following error message when I try to use the "select data for microstate analysis" step in the Microstate analysis toolbox.

EEGLAB error in function pop_micro_selectdata() at line 172:
License checkout failed.
License Manager Error -15
Unable to connect to the license server.
Check that the network license manager has been started, and that the client machine can communicate with the license server.
Troubleshoot this issue by visiting:
httos://www.mathworks.com/support/Ime/15
Diagnostic Information:
Feature: Signal Toolbox
License path:
Licensing error: - 15,0.
If you think this is a bug, please submit a detailed description of how to reproduce the problem (and upload a small dataset) at httos://github.com/scc/eeglab/issues

#### Steps to Reproduce
File>load existing datasets from
https://drive.google.com/drive/folders/1tflOWA9-oFqZGlgvHBnPJ0YID2Cm1_qR?usp=sharing
Tools>Microstate analysis toolbox>select data for microstate analysis
[image]

I have signal processing toolbox installed so that's not the issue. See below:

MATLAB Version: 23.2.0.2515942 (R2023b) Update 7
MATLAB License Number: unknown
Operating System: Microsoft Windows 10 Enterprise Version 10.0 (Build 19044)
Java Version: Java 1.8.0_202-b08 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
MATLAB Version 23.2 (R2023b)
Simulink Version 23.2 (R2023b)
Data Acquisition Toolbox Version 23.2 (R2023b)
Deep Learning Toolbox Version 23.2 (R2023b)
Signal Processing Toolbox Version 23.2 (R2023b)
Statistics and Machine Learning Toolbox Version 23.2 (R2023b)

arnodelorme, 2024-03-04:
You need to purchase the signal processing toolbox from the Mathworks or try on Octave. Best

closed 2024-03-04 by arnodelorme

Githubwl66, 2024-03-04:
> You need to purchase the signal processing toolbox from the Mathworks or try on Octave. Best

Could you advise how to purchase the signal processing toolbox from Mathworks? I don't quite understand why it was installed but not purchased.