sccn/eeglab #742: EEGLAB error in function celluniquelegacy

Zrn1213, 2024-03-06:
In EEGLAB version 2023.1, when attempting to reject data manually, an error occurred, stating "EEGLAB error in function celluniquelegacy" after clicking on the menu. How to solve this question?

arnodelorme, 2024-03-06:
Would you mind copying the exact error message. Is the function missing? What MATLAB version are you using. Thank you

Zrn1213, 2024-03-07:
Thanks for your letter. I have solved the problem. I found I had some mistakes in my code which caused this problem. Thanks again.

Wrong code:
for i=1:length(EEG.event)        
if double(strcmp(EEG.event(i).type,'boundary'))==1
   EEG.event(i).type=0;
end
   end

Ruonan Zhao
March 7, 2024

arnodelorme, 2024-03-07:
Good

closed 2024-03-07 by arnodelorme