sccn/eeglab #757: readeetraklocs chokes on head shape

Allexxann, 2024-04-15:
xensor .elc files can include head shape, and this sometimes breaks the function. Or maybe all the times, I'm not sure.
I dug around, in the process I've found mentions of readasalocs, but apparently it was discontinued at some point.

The offending line is
``labels    = locs(indlabels+1:end,:);``
this reads the entirety of the file til the end. If the labels are the last line - you're lucky. Otherwise - not so much.

``labels    = locs(indlabels+1,:);``
This reads only the labels.

There was also a for-if loop, and I replaced this out of principle.

	locs  = loadtxt( filename );    
    % get label names
    % ---------------
	
	indlabels=find(strcmpi(locs(:,1),  'Labels'));
	indpos=find(strcmpi(locs(:,1), 'Positions'));
	
    if isempty(indpos) || isempty(indlabels)
        error('Could not find ''Labels'' or ''Position'' tag in electrode file');
	end
	
    labels    = locs(indlabels+1,:);
    
    % get positions
    % -------------
    if strcmp(locs(indpos+1,2),':')
        positions = locs(indpos+1:indlabels-1,3:5);
    else
        positions = locs(indpos+1:indlabels-1,1:3);
    end

arnodelorme, 2024-04-16:
Do you have an .elc file we could use for testing?

Allexxann, 2024-04-17:
[ELC.zip](https://github.com/sccn/eeglab/files/15010068/ELC.zip)
Directly from xensor without any tampering.

arnodelorme, 2024-07-02:
This has been fixed in 82d3f5696

closed 2024-07-02 by arnodelorme