sccn/eeglab #762: Discrepancy between std_erspplot plot and exported data

sarahakhras, 2024-04-17:
Hi, 

My aim is to extract ERSP data points within a specific bandwidth and timeframe so I can run some stats. I was able to extract the erspdata by calling std_erspplot and looping over subject, condition, erspfreqs, and ersptimes. However, when double-checking the output against std_erspplot ERSP plot with matching params the power values didn't quite match. Any ideas on why there's a discrepancy? 

For the mat code, I used the suggested code found on the EEGLAB Wiki Group-level analysis for extracting ERSP data with some computing enhancements. 

Thanks!

arnodelorme, 2024-04-17:
Using the Sternberg tutorial study, left is the plot from the GUI, right is the figure created using the code below. Seems to match for me (different color scales).

```
[STUDY,ersp] = std_erspplot(STUDY,ALLEEG,'channels',{'FP1'}, 'design', 2);
figure; imagesc(mean(ersp{1}, 4));
```

[image]

closed 2024-04-17 by arnodelorme

sarahakhras, 2024-04-17:
Thanks for your response! I think the common baseline subtraction was the main issue when comparing.