sccn/eeglab #765: Missing 'ft_prepare_neighbours' function

marcelzwiers, 2024-04-23:
### Description

I'm running a script that calls `std_prepare_neighbors()`, which then generates the following error:

```matlab
[..]
Removing 3 channel(s)...
Event resorted by increasing latencies.
Unrecognized function or variable 'ft_prepare_neighbours'.

Error in std_prepare_neighbors (line 123)
            cfg.neighbors = ft_prepare_neighbours(tmpcfg, tmpcfg2);

Error in ERP_Core_WB (line 89)
    [~,~,AvgChanlocs] = std_prepare_neighbors(STUDY, ALLEEG, 'force', 'on');
```

I searched through the eeg lab repository but could not find that function. Is it somehow missing or should I install Fieldtrip as a dependency?

#### Versions

|   |   |
|----| ---- | 
| OS version     | CentOS 7.4 |
| Matlab version | 2023b |
| EEGLAB version | 2024.0 (i.e. current) |

robertoostenveld, 2024-04-24:
FieldTrip definitely includes that function. FieldTrip could be installed as separate package and added to the path, which would solve it. There is also the possibility to install FieldTrip as a plugin in EEGLAB. I am curious as to what Cyril considers the "clean" solution.

arnodelorme, 2024-04-24:
Yes, make sure you have fieldtrip installed.

closed 2024-04-24 by arnodelorme