sccn/eeglab #766: Group level analysis: different channels per frame error

droe11, 2024-04-23:
Hi All, 

I am trying to perform my first group level analysis on some of my data to familiarize myself with the technique for when I have a full dataset. However, when trying to perform the group level analysis using the EEGLAB gui I am getting an error code. I have preprocessed my data at the single subject level and I have then precomputed the channel measures. 

But then when I try to do visualization and statistical analyses I get the following error message: 
number of elements must not change. Use [] as one of the size inputs to automatically calculate the appropriate size for that dimension., this is not a bug (error occurred in function statcond() at line 258)

It says the epochs are consistent but the channels per frame are different per participant. However, on the EEGLAB tutorial it says that it can be performed with different channels per frame. 

Any help would be greatly appreciated. 

Thank you EEGLAB community. 

Dominic

arnodelorme, 2024-06-18:
Would you mind to let us which part of the tutorial you are following?

arnodelorme, 2024-07-02:
Closing this report for now, but please reopen if you can find which part of the tutorial creates the error.

closed 2024-07-02 by arnodelorme