sccn/eeglab #768: Error: std_readdata in Line 211 (Unequal number of sessions)

zaeemhadi, 2024-04-29:
Dear EEGLAB team,

I was loading 2 participants with 30 set files (5 runs x 6 conditions) and one participant with 24 set files (4 runs x 6 conditions).

I kept getting the error "Cannot process components from different ICA decompositions of the same subjects". I made sure multiple times that the ICA weights are same per participant, however, it didn't fix the issue.

In the end, commenting out lines 210-212 fixed the issue. I was wondering if these lines can be safely commented out?

Could the error message be made more informative, as unequal number of runs shouldn't be a problem.

Best wishes,
Zaeem

arnodelorme, 2024-04-29:
Probably not. Can you try "std_findsameica(ALLEEG)?" In the example below, there are 6 datasets and 3 subjects. Subject 1 is datasets 1 and 2, subject 2 is datasets 3 and 4, and subject 3 is datasets 5 and 6. 

The function you are using will return an error if there is not the same number of elements for each subject (one subject has more than others). Let us know.

```matlab
std_findsameica(ALLEEG)

ans =

  1×3 cell array

    {[1] [2]}    {[3] [4]}    {[5] [6]}
```

zaeemhadi, 2024-04-30:
As above, I loaded 3 participants with 30 set files (5 runs x 6 conditions) and one participant with 24 set files (4 runs x 6 conditions). Do I have to drop the subject with unequal datasets?

std_findsameica(ALLEEG) returns below

`ans =

  1×4 cell array

  Columns 1 through 2

    {1×30 double}    {1×30 double}

  Columns 3 through 4

    {1×24 double}    {1×30 double}`

arnodelorme, 2024-04-30:
In your case, I think it is fine to comment the lines in the code. I was trying to find which function returned the error. Where did you comment the lines?

zaeemhadi, 2024-04-30:
Oh. I got the error when I was trying to plot ERSPs from edit/plot component clusters. 
The error appeared while the command window was printing

> Reading subjects' data or looking up measure values in EEGLAB cache

I commented the lines 210-212 inside the function "std_readdata.m" (inside folders functions -> studyfunc).

arnodelorme, 2024-04-30:
This function was changed in EEGLAB 2024.0. I do not think it would trigger the error now. Would you mind to give it a try?

zaeemhadi, 2024-04-30:
I am using the EEGLAB version 2024.0. The error occurred in this version.

arnodelorme, 2024-04-30:
I do not see the relevant lines at 210-212. Attaching the latest function. Would you mind to try it?

[std_readdata.m.zip](https://github.com/sccn/eeglab/files/15156383/std_readdata.m.zip)

zaeemhadi, 2024-04-30:
Thanks for sending this. It worked perfectly fine with this function.

Just to double check, I tried redownloading latest EEGLab for matlab from below link
[https://sccn.ucsd.edu/eeglab/downloadtoolbox.php](url)

However, the std_readdata.m function that comes with it doesn't seem to be updated.

Best wishes

arnodelorme, 2024-04-30:
Yes, the function I shared will be in 2024.1 in a couple of months. You can access it on Github but not through that link. Thank you

closed 2024-04-30 by arnodelorme