sccn/eeglab #771: bug in calling binica (on Linux)

jnvandermeer, 2024-05-13:
If you call binica from the gui,

then icadefs is called, which sets ICABINARY:

`ICABINDARY = fullfile(eegpab_p, 'functions','supportfiles','ica_linux')`

but then, binica.m is called (which then calls the ICABINARY), but there, it is written:

line 115:
`icadefs`
line 117:
`ICABINDARY = fullfile(eegpab_p, 'functions','supportfiles','ICABINARY')`

... which effectively makes ICABINARY unfindable.

If I change binica.m, on line 117, and replace ICABINARY with 'ica_linux', it works again.

I *think* this is a bug in eeglab, and may only be with linux, and not (very) often noticed because everybody just uses runica.m

arnodelorme, 2024-05-13:
I think this issue has been fixed because icadefs.m no longer has the code setting the full path

```
ICABINDARY = fullfile(eegpab_p, 'functions','supportfiles','ica_linux')
```

jnvandermeer, 2024-05-13:
Hi Arnoud, you are right! 

It was ERPLAB10.04, the eeglab plugin, that also contained and icadefs.m file. Because I added this plugin, the icadefs of ERPLAB was used, which lives in `plugins/ERPLAB10.04/studio_functions/Functions/EEGLAB/Scalp/icadefs.m`. 

So, this probably better belonged to the ERPLAB repository.

arnodelorme, 2024-05-13:
Interesting point, one more reason to retire icadefs.m (which I was planning to do). I will implement a fix.

arnodelorme, 2024-07-02:
@guanghuizhang0328 and @davidg622, how can we avoid the problem above?

arnodelorme, 2025-02-07:
This was fixed in c8badde.

closed 2025-02-07 by arnodelorme