sccn/eeglab #772: Warning when using pop_rmdat()

hannaringer, 2024-05-17:
### Description

When removing continuous data around specific events using pop_rmdat(), I keep getting the following warning: "Warning: Colon operands must be real scalars. This warning will become an error in a future release." 
Currently it is just a warning and the function seems to work alright nevertheless. I just wanted to still report it since it may be become a problem in future releases. 
I tried to track down where exactly the warning comes from, and I think I indeed found it in eegrej.m: It seems that in line 139 an indication of the correct dimension of "size(regions)" is missing. Accordingly, the warning disappears when I replace 
"for iRegion2=iRegion1+1:size(regions)" with "for iRegion2=iRegion1+1:size(regions, 1)" in line 139 of eegrej.m

I hope this is of any help to anyone encountering the same warning. 

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | MacOS Sonoma 14.2.1 |
| Matlab version | 2024a |
| EEGLAB version | 2024.0 |

arnodelorme, 2024-05-17:
Hi Hanna, It could be that you have a non-integer sampling frequency, like 250.23 Hz. Can you check?

hannaringer, 2024-05-17:
Hi Arno, actually I don't - my sampling frequency is 500 Hz.

arnodelorme, 2024-05-17:
This is strange. Can you upload a dataset on which I can run pop_rmdat? We will check. Compress it as zip and drag and drop here. Thank you.

hannaringer, 2024-05-17:
Thanks for looking into this. Here is an excerpt of a dataset of an exemplary participant:
[example.zip](https://github.com/sccn/eeglab/files/15345082/example.zip)

arnodelorme, 2024-05-17:
Thank you Hanna. There was indeed a minor glitch, although the function was working properly. This has been fixed in 02a306cce

closed 2024-05-17 by arnodelorme

hannaringer, 2024-05-20:
Great, thank you!